hypriot/image-builder-rpi #251: Add automated build on Docker Hub

breandan, 2018-06-16:
I noticed there is a [`Dockerfile`](https://github.com/hypriot/image-builder-rpi/blob/master/Dockerfile), but nothing at https://hub.docker.com/r/hypriot/image-builder-rpi/. It would be convenient to have an automated build to base an image on top of this. Thanks!

StefanScherer, 2018-06-16:
There is a `Dockerfile`, but we do not build a Docker image in this repo. The Dockerfile is a dev environment to build the Raspberry Pi SD card image that you have to flash onto a SD card.

Have a look at https://blog.hypriot.com/post/making-of-hypriotos/ - a little outdated as we moved from TravisCI to CircleCI, but the build steps are almost the same.

So we have automated builds, you can even fork our repo and attach the fork to CircleCI to have a forked build pipeline with small adjustments and configuration.

closed 2018-06-16 by StefanScherer

breandan, 2018-06-16:
I am building a Docker image to build a custom Hypriot image preloaded with another Docker image following https://github.com/hypriot/image-builder-rpi/issues/244#issuecomment-392195067. When the user runs `docker run --privileged -it my/image`, it will flash the custom Hypriot image (containing my Docker image) onto an SD card. I would like this image to be built on the Docker Hub. Any advice?

StefanScherer, 2018-06-16:
Wow, a great challenge. But I don't have any advice here.
I don't know if Docker Hub automated builds are able to create the SD card image. We use guestfish and with this we had trouble to run it in Docker. We had to move to a Vagrant box or a machine type build at CircleCI.