hypriot/image-builder-rpi #253: Hypriot to be based on Alpine ?

nsteinmetz, 2018-06-29:
Hello there,

With Alpine 3.8 having support of RPI3, I was wondering if you are to release a Hypriot OS version, based on Alpine instead of raspbian ?

https://alpinelinux.org/posts/Alpine-3.8.0-released.html

```
New features and noteworthy new packages

    Support netboot on all architectures
    Add arm64 (aarch64) Raspberry Pi image
    Add support for Raspberry Pi 3 Model B+
    Support ISO image on s390x (KVM installation)
    End of support for hardened kernel (unofficial Grsecurity)
    Support for Crystal language

Significant updates

    Linux 4.14
    Go 1.10
    Node.js 8.11 (LTS)
    Rust 1.26
    Ruby 2.5
    PHP 7.2
    ghc 8.4
    OCaml 4.06
    R 3.5
    JRuby 9.2
```

DieterReuter, 2018-06-30:
That's really a great idea! But not sure if I will have the time for...

nsteinmetz, 2018-07-01:
Perhaps I can work on it after summler break with some mentoring or minimal set of instructions.

74cmonty, 2018-07-05:
Just a comment: the new Alpine release 3.8 is "designed for RPI 1,2 and 3".

ulm0, 2018-07-22:
Wow, this might be a really nice flavor!

toddbu, 2018-09-01:
I'd urge caution jumping in on this. About a year ago and after many days of debugging, I was not able to get node.js to run in an Alpine container running on an Alpine host. It wouldn't even start. I use Alpine exclusively for all my deployments and never had this issue with any other Docker host (Ubuntu, CoreOS, or Raspbian)

StefanScherer, 2019-01-06:
As we do not have time for such adventures  I'll close the ticket.

closed 2019-01-06 by StefanScherer

nsteinmetz, 2019-01-06:
Indeed - lack of time also on my side for this challenge - sorry and may be next time ;-)