hypriot/image-builder-rpi #257: Can't open docker

Jacobh2, 2018-09-21:
Hey!

I've configured a cloud-init file to login to my private docker image repo using the `docker login` command.

I've placed the command under the `runcmd` and it looks something like this:

```
runcmd:
  - 'systemctl restart avahi-daemon'
  - 'sudo -u client /bin/sh docker login -u _json_key --password-stdin https://eu.gcr.io < /tmp/client/gcloudauth.json'
```

But when I then ssh into the pi and check the log (`/var/log/cloud-init-output.log`) all I see is:
`/bin/sh: 0: Can't open docker`

Is the docker daemon not started yet? How can I run one-time commands on the pi that involves the docker daemon?

StefanScherer, 2018-09-21:
I don't know exactly when Docker is running, but I would expect it should be running at that point.
Or you can add another systemctl command to be sure docker is running.

StefanScherer, 2019-09-16:
Closing due to inactivity

closed 2019-09-16 by StefanScherer