hypriot/image-builder-rpi #258: cannot boot 1.9.0 image on RPI 1

eyakcn, 2018-09-21:
I see the description that 1.9.0 release works on all Raspberry Pi models (1, 2, 3, 3 B+, Zero, Zero W & Compute Module) as the Docker Engine 18.04.0-ce is built for ARMv6

My Raspberry PI was bought in 2014, maybe it's RPI 1.

Anyway, after the image flashed into the SD card, it can not boot. The board never light on except the power. I tried to give it HDMI output, but nothing happens.

Here is the user-data I used:

#cloud-config
# vim: syntax=yaml
#

# The current version of cloud-init in the Hypriot rpi-64 is 0.7.6
# When dealing with cloud-init, it is SUPER important to know the version
# I have wasted many hours creating servers to find out the module I was trying to use wasn't in the cloud-init version I had
# Documentation: http://cloudinit.readthedocs.io/en/0.7.9/index.html

# Set your hostname here, the manage_etc_hosts will update the hosts file entries as well
hostname: rasp512
manage_etc_hosts: true

# You could modify this for your own user information
users:
  - name: peter
    gecos: "Peter SONG"
    sudo: ALL=(ALL) NOPASSWD:ALL
    shell: /bin/bash
    groups: users,docker,video,input
    plain_text_passwd: xxxxxxxx
    lock_passwd: false
    ssh_pwauth: true
    chpasswd: { expire: false }

# # Set the locale of the system
locale: "zh_CN.UTF-8"

# # Set the timezone
# # Value of 'timezone' must exist in /usr/share/zoneinfo
timezone: "Asia/Shanghai"

# # Update apt packages on first boot
# package_update: true
# package_upgrade: true
# package_reboot_if_required: true
package_update: false
package_upgrade: false

# # Install any additional apt packages you need here
# packages:
#  - ntp

# # WiFi connect to HotSpot
# # - use `wpa_passphrase SSID PASSWORD` to encrypt the psk
# write_files:
#   - content: |
#       allow-hotplug wlan0
#       iface wlan0 inet dhcp
#       wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
#       iface default inet dhcp
#     path: /etc/network/interfaces.d/wlan0
#   - content: |
#       country=de
#       ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
#       update_config=1
#       network={
#       ssid="YOUR_WIFI_SSID"
#       psk="YOUR_WIFI_PASSWORD"
#       proto=RSN
#       key_mgmt=WPA-PSK
#       pairwise=CCMP
#       auth_alg=OPEN
#       }
#     path: /etc/wpa_supplicant/wpa_supplicant.conf
write_files:
  - content: |
      {
        "registry-mirrors": ["https://registry.docker-cn.com"]
      }
    path: /etc/docker/daemon.json

# These commands will be ran once on first boot only
runcmd:
  # Pickup the hostname changes
  - systemctl restart avahi-daemon
  - systemctl restart docker

#  # Activate WiFi interface
#  - ifup wlan0

StefanScherer, 2018-09-22:
Hey @eyakcn 
I tried your cloud-init yml file and flashed a 1.9.0 SD image and booted a Raspberry Pi 1 from 2012. 
On HDMI output you should see the resize filesystem dialog, then the pi reboots and logs its boot steps and cloud-init steps.
As you cannot see anything please check if you have flashed the SD image correctly.
There should be a FAT partition after flashing it and that partition is visible on a Windows or Mac computer.

You can also check you cloud-init file which is a YAML and sometimes hard to get the indenation right.
On my Mac I could use this line to verify the YML file.

```
ruby -e "require 'yaml';YAML.load_file('./issue258.yml'); puts 'YAML is ok'"
```

or use an online yaml validator https://coreos.com/validate/

The correct indented cloud-init file that works for me is:

```yaml
#cloud-config

hostname: rasp512
manage_etc_hosts: true

users:
  - name: peter
    gecos: "Peter SONG"
    sudo: ALL=(ALL) NOPASSWD:ALL
    shell: /bin/bash
    groups: users,docker,video,input
    plain_text_passwd: xxxxxxxxxx
    lock_passwd: false
    ssh_pwauth: true
    chpasswd: { expire: false }

locale: "zh_CN.UTF-8"

timezone: "Asia/Shanghai"

package_update: false
package_upgrade: false

write_files:
  - content: |
      {
      "registry-mirrors": ["https://registry.docker-cn.com"]
      }
    path: /etc/docker/daemon.json

runcmd:
  - 'systemctl restart avahi-daemon'
  - 'systemctl restart docker'
```

muelli, 2019-08-09:
This can be closed, I guess.

StefanScherer, 2019-09-16:
Closing due to inactivity

closed 2019-09-16 by StefanScherer