hypriot/image-builder-rpi #263: Investigate Balena as alternative for Docker runtime

bltavares, 2018-10-22:
Hi there,

Firs of all, thanks for the amazing project, it has proven to be quite valuable as it is right now!
This is more of a question for investigation and feedback on the idea, than an issue itself.

I've came across [balena](https://www.balena.io/), which claims to be a compatible Docker engine focused for Embedded and small devices:

> Balena is a new container engine purpose-built for embedded and IoT use cases and compatible with Docker containers.
>
> Based on Docker’s Moby Project, balena supports container deltas for 10-70x more efficient bandwidth usage, has 3.5x smaller binaries, uses RAM and storage more conservatively, and focuses on atomicity and durability of container pulling.

It does remove some of the featuers, such as Docker Swarm tho.

I was wondering if it would be of the interest of hypriot to also produce an alternative engine, and if so, how we could help with this.

StefanScherer, 2018-10-22:
Hi @bltavares 

Thank you for the kind words. I've seen Balena last year at Moby Summit and it has some nice optimizations for smaller devices.

At the moment we do not plan to add more images.

A recommended way to produce an image for the community is to just fork this repo and activate the CircleCI build for it. Then you have a first build pipeline that produces build artifacts for each git commit. So you can adjust the software to be installed in the image as you need it.
If you want to draft releases you have to add the GITHUB_TOKEN variable to your forked CircleCI build and maybe adjust the deploy steps to release to your own fork. That's more or less everything to clone our pipeline.

closed 2018-11-01 by StefanScherer

bltavares, 2018-11-01:
Sorry for not responding. Thanks for the suggestion.

I'll give it a try and report back, if you are interested :)