hypriot/image-builder-rpi #264: Problems with cgroup with hypriot 1.9 and kubeadm

dgem, 2018-11-06:
Hi,
I'm unable to kubeadm to pass the system verification steps. Using a self built kubeadm:
```
kubeadm init --pod-network-cidr 10.244.0.0/16
```
I get the following pre-flight errors:
```
[init] using Kubernetes version: v1.12.2
[preflight] running pre-flight checks
[preflight] The system verification failed. Printing the output from the verification:
KERNEL_VERSION: 4.14.34-hypriotos+
CONFIG_NAMESPACES: enabled
CONFIG_NET_NS: enabled
CONFIG_PID_NS: enabled
CONFIG_IPC_NS: enabled
CONFIG_UTS_NS: enabled
CONFIG_CGROUPS: enabled
CONFIG_CGROUP_CPUACCT: enabled
CONFIG_CGROUP_DEVICE: enabled
CONFIG_CGROUP_FREEZER: enabled
CONFIG_CGROUP_SCHED: enabled
CONFIG_CPUSETS: not set
CONFIG_MEMCG: enabled
CONFIG_INET: enabled
CONFIG_EXT4_FS: enabled
CONFIG_PROC_FS: enabled
CONFIG_NETFILTER_XT_TARGET_REDIRECT: enabled (as module)
CONFIG_NETFILTER_XT_MATCH_COMMENT: enabled (as module)
CONFIG_OVERLAY_FS: enabled (as module)
CONFIG_AUFS_FS: not set - Required for aufs.
CONFIG_BLK_DEV_DM: enabled (as module)
DOCKER_VERSION: 18.06.1-ce
DOCKER_GRAPH_DRIVER: overlay2
OS: Linux
CGROUPS_CPU: enabled
CGROUPS_CPUACCT: enabled
CGROUPS_CPUSET: missing
CGROUPS_DEVICES: enabled
CGROUPS_FREEZER: enabled
CGROUPS_MEMORY: enabled
[preflight] Some fatal errors occurred:
	[ERROR SystemVerification]: unexpected kernel config: CONFIG_CPUSETS
	[ERROR SystemVerification]: missing cgroups: cpuset
[preflight] If you know what you are doing, you can make a check non-fatal with `--ignore-preflight-errors=...`
```
The bootcmd.txt file contains:
```
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 cgroup_cpuset=1 cgroup_enable=cpuset cgroup_enable=memory swapaccount=1 elevator=deadline fsck.repair=yes rootwait
```
Is it possible the latest hypriot image wasn't built with the correct options ?

dbwest, 2018-11-21:
I'm trying `kubeadm init --pod-network-cidr 10.244.0.0/16 --ignore-preflight-errors=SystemVerification`...

StefanScherer, 2019-09-16:
Closing due to inactivity

closed 2019-09-16 by StefanScherer