hypriot/image-builder-rpi #268: OS updates on Raspberry Zero W breaks dockerd

guidoffm, 2018-11-25:
- Freshly installed Hypriot 1.9 on SD card.
- Booted the system for the first time
- Ran ```sudo apt update && sudo apt dist-upgrade -y```
- Installation says:

```
update-alternatives: using /usr/bin/dockerd-ce to provide /usr/bin/dockerd (dockerd) in auto mode
Job for docker.service canceled.
docker.service couldn't start.
```

- Tried to run dockerd manually. -> Segmentation fault

guidoffm, 2018-11-25:
Syslog reads:

```
Nov 25 16:59:00 rspz1 systemd[1]: Starting containerd container runtime...
Nov 25 16:59:00 rspz1 systemd[1]: Started containerd container runtime.
Nov 25 16:59:02 rspz1 systemd[1]: containerd.service: Main process exited, code=dumped, status=4/ILL
Nov 25 16:59:02 rspz1 systemd[1]: containerd.service: Unit entered failed state.
Nov 25 16:59:02 rspz1 systemd[1]: containerd.service: Failed with result 'core-dump'.
Nov 25 16:59:53 rspz1 systemd[1]: Reloading.
```

guidoffm, 2018-11-25:
I rolled back the docker-ce installation to ```18.06.1-ce``` and marked the package as hold to prevent further updates.

StefanScherer, 2018-12-02:
Try the new v1.10.0-rc1 SD image from the github releases. A dist-upgrade is not our focus

guidoffm, 2018-12-02:
Thanks Stefan, I will give it a try.

closed 2018-12-02 by guidoffm