hypriot/image-builder-rpi #270: cloud-init users ssh_import_id fails because of missing ssh-import-id

danmilon, 2018-12-13:
```
2018-04-28 18:57:20,507 - util.py[DEBUG]: Running module ssh-import-id (<module 'cloudinit.config.cc_ssh_import_id' from '/usr/lib/python3/dist-packages/cloudinit/config/cc_ssh_import_id.py'>) failed
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cloudinit/stages.py", line 776, in _run_modules
    freq=freq)
  File "/usr/lib/python3/dist-packages/cloudinit/cloud.py", line 54, in run
    return self._runners.run(name, functor, args, freq, clear_on_fail)
  File "/usr/lib/python3/dist-packages/cloudinit/helpers.py", line 187, in run
    results = functor(*args)
  File "/usr/lib/python3/dist-packages/cloudinit/config/cc_ssh_import_id.py", line 86, in handle
    raise elist[0]
  File "/usr/lib/python3/dist-packages/cloudinit/config/cc_ssh_import_id.py", line 79, in handle
    import_ssh_ids(import_ids, user, log)
  File "/usr/lib/python3/dist-packages/cloudinit/config/cc_ssh_import_id.py", line 107, in import_ssh_ids
    raise exc
  File "/usr/lib/python3/dist-packages/cloudinit/config/cc_ssh_import_id.py", line 104, in import_ssh_ids
    util.subp(cmd, capture=False)
  File "/usr/lib/python3/dist-packages/cloudinit/util.py", line 1850, in subp
    cmd=args)
cloudinit.util.ProcessExecutionError: Unexpected error while running command.
Command: ['sudo', '-Hu', 'danmilon', 'ssh-import-id', 'gh:danmilon']
```

`ssh-import-id` is missing. When it's installed it works fine.

Should it be installed by default, or should I add it in the list of packages in the userdata file? In that case, does it install packages before or after setting up users?.

Thanks!

StefanScherer, 2018-12-13:
Thanks. Is already fixed with https://github.com/hypriot/image-builder-rpi/commit/cc8d4191e961b540520f245c21c11529f13e03c5 and will be part of our next SD card release.

closed 2018-12-13 by StefanScherer

danmilon, 2018-12-14:
Oh I didn't notice. Thanks!