hypriot/image-builder-rpi #271: headsup: docker arm6 broken

FlorianHeigl, 2018-12-15:
Hi,

while I'm building our xmas gifts... I just ran into some issue, they track it at docker here:
https://github.com/moby/moby/issues/38175

Solution: On ARMv6 one needs to downgrade docker to 18.06.1.
This is done by running ```# apt-get install docker-ce=18.06.1~ce~3-0~raspbian``` and gave me a working docker daemon.

----

What happens is they got a go issue and all parts of docker coredump :)
```
# systemctl start docker
Job for docker.service failed because a fatal signal was delivered causing the control process to dump core.
See "systemctl status docker.service" and "journalctl -xe" for details.
HypriotOS/armv6: root@rainbow in ~
# systemctl status docker
● docker.service - Docker Application Container Engine
   Loaded: loaded (/lib/systemd/system/docker.service; enabled; vendor preset: e
   Active: inactive (dead) (Result: core-dump) since Sat 2018-12-15 02:45:39 UTC
     Docs: https://docs.docker.com
  Process: 2327 ExecStart=/usr/bin/dockerd -H unix:// (code=dumped, signal=SEGV)
 Main PID: 2327 (code=dumped, signal=SEGV)
      CPU: 462ms

Dec 15 02:45:39 rainbow systemd[1]: docker.service: Unit entered failed state.
Dec 15 02:45:39 rainbow systemd[1]: docker.service: Failed with result 'core-dum
Dec 15 02:45:39 rainbow systemd[1]: Stopped Docker Application Container Engine.
```

I'm not sure how I'll now adjust the image to include an older version, likely just run the image builder instead of just ```# flash```

In hindsight I find it really funny I expected I would just build those raspis using all known parts and it'd be all fine. Like, really? How did I think that! :-)

hjacobs, 2018-12-28:
I had the same problem (Docker core-dump) today on Raspi 2 trying out v1.10.0-rc1 (#273). To make it work, I had to downgrade to 18.06.1:

```
apt-get update
apt-get install docker-ce=18.06.1~ce~3-0~raspbian
```

nickcj931, 2019-01-03:
@FlorianHeigl thanks for posting this, fix has worked! 
If only things were simple I'm trying to get my Zigbee to MQTT to work spent hours just trying to build the zero hahaha

rjelbert, 2019-01-06:
I have just tried to run v1.10.0-rc1 on three different Raspberry Pi types. I had two different issues.
Here is the complete list and the results:
Raspberry Pi model B+ v1.2 (Docker failed to start - had to downgrade)
Raspberry Pi 2 model B v1.1 (Docker seemed to start/work but Registry refused to run. Had to downgrade. This pi is an arm7)
Raspberry Pi 3 model B+ (it worked)
Hope this helps

StefanScherer, 2019-01-06:
Probably we should ship Docker 18.06.1 in the 1.10.0 release. I'll update the build and trigger another pre-release.

closed 2019-01-06 by StefanScherer

StefanScherer, 2019-01-07:
There is a pre-release 1.10.0-rc2 with Docker 18.06.1.

PR #275 (2019-01-06): Downgrade to 18.06.1 to support all RPi models
Downgrade from 18.09.0 to 18.06.1 to support all Raspberry Pi models.

Fixes #271