hypriot/image-builder-rpi #276: Accessing GPIO in Swarm mode - a suggestion / proposal at SD card flashing stage

rjelbert, 2019-01-07:
The Raspberry Pi is a hardware hackers tool dream so it seems at odds that Hypriot & Docker do not support the easy reading of sensors or IoT devices connected to the Pi when in swarm mode.

I am proposing / requesting an installation time option to enable an execution mode that will allow the equivalent of privileged containers when deployed normally OR swarm.

I am suggesting that Hypriot is adapted so if a flag is set when the image is flashed to the SD card, ALL containers are run as if in privilege mode on all nodes. This will allow for simple, direct and high speed access to GPIO and I2C and other busses. 

I know this could allow for undesirable consequences if misused but most people using this distribution will be the owners of the cluster and desperate to connect lots of cool sensors, motors and other devices - something they can not easily do at the moment when using swarm. It’s our gear and we are grownups so please consider offering this option (please).

I’d be interested in your feedback on my suggestion because it builds on and supports the unique positioning of the Hypriot environment for RPi and the hardware hacker community behind it.

Richard

P.S. 

I came to my proposal above after researching the suggested ways to access GPIO in a Swarm. Note that while possible, hardware hackers are looking for simplicity not complexity. Suggested solutions as of 7th January 2019 can be summarised as:

Use Kubernetes with Hypriot & Docker for container management instead of Swarm because it seems to support privileged containers. The downside is that it is complex and requires the reader to install and learn another complex technology potentially just to flash LEDs on a cluster.
[https://blog.hypriot.com/post/setup-kubernetes-raspberry-pi-cluster/](https://blog.hypriot.com/post/setup-kubernetes-raspberry-pi-cluster/)

Use sysfs with Hypriot & Docker instead of direct access to /dev/gpiomem. There is a clear tutorial linked below by Alex Ellis. Unfortunately this type of access is slow compared to direct access and hardware people like to have as low level access to the device as possible.
[https://blog.alexellis.io/gpio-on-swarm/](https://blog.alexellis.io/gpio-on-swarm/ )
[https://www.appcelerator.com/blog/2018/11/running-api-builder-on-raspberry-pi-cluster-using-docker-swarm-part-3/](https://www.appcelerator.com/blog/2018/11/running-api-builder-on-raspberry-pi-cluster-using-docker-swarm-part-3/) 

Use Ansible in combination with Hypriot & Docker to access the GPIO / hardware. Ansible is a flexible IT automation engine. Again this is another layer to add the Hypriot / Docker engine to access GPIO but it also complicates the task.
[https://www.ansible.com/](https://www.ansible.com/) 
[https://github.com/tIsGoud/docker-swarm-on-a-raspberry-pi-cluster )](https://github.com/tIsGoud/docker-swarm-on-a-raspberry-pi-cluster )

One final solution might be to use an alternative to Hypriot to get native GPIO access but there is a commercial model to contend with. Looks neat but Hypriot or Raspbian are free!
[https://www.balena.io/docs/learn/getting-started/raspberrypi3/nodejs/](https://www.balena.io/docs/learn/getting-started/raspberrypi3/nodejs/)

In summary, these official work arounds add complexity and in some cases restrict or slow down proper hardware access to the RPi hardware.

StefanScherer, 2019-01-08:
Swarm mode does not provide privileged services https://github.com/moby/moby/issues/24862 there seems to be some work in progress to add specificy capabilities to swarm mode.

Ansible is just an automation tool, you can configure GPIO without it as well.

Balena is a different container runtime, which should be open source. The product by resin.io is of course a commercial product which saves you a lot of manual steps.

One of our students is currently working on a Blinkt demo in swarm mode with a trick by another Docker Captain. He uses a per-node privileged container that is created from a swarm mode service to access GPIO. I hope we can share more in near future.

I've used the /sys for my DockerCon 2017 talk in Austin https://www.slideshare.net/stefscherer/swarm-2-go-build-a-portable-multiarch-data-center-with-pi-and-up-nodes
Some of the commands are listed in this repo https://github.com/plossys/blinkt

rjelbert, 2019-01-08:
Thanks. I'd be interested in hearing about the trick you mentioned. Happy to test it. I need to directly control GPIO and I2C from a swarm container.

My proposal above was designed as a quick solution and to make as few changes to the code as possible and to force all Docker containers to run with root privileges and therefore be able to access GPIO directly. 

I have also seen that Ciro S. Costa has listed the instructions to fork Docker to add the privileged switch to swarm. As a "proper" solution, this might make a nice addition to Hyperiot... 
[ https://ops.tips/blog/adding-privileged-to-docker-swarm-mode/](https://ops.tips/blog/adding-privileged-to-docker-swarm-mode/)

StefanScherer, 2019-01-08:
A very good blog post how to fix it by compiling the feature into docker binaries😎
We at Hypriot we did great effort to move all changes to upstream and from then on only consume new packages. We‘re not keen on start compiling a own version again.

seal-prw, 2019-01-08:
> Thanks. I'd be interested in hearing about the trick you mentioned. Happy to test it. I need to directly control GPIO and I2C from a swarm container.
>
> My proposal above was designed as a quick [...]

As Docker Swarm does not support privileged mode yet, the workaround to this problem that I tried out was starting a privileged container from inside of the software program by using Docker Engine api. This resulted into creation of one additional container with privileged access per node and this container then handles all the tasks which requires privileged access. The software program checks whether the container is already exists, if not then only new container is created.  For reference please have a look at this git repo which contains blinkt swarm cluster demo:
https://github.com/seal-prw/tinycloud.git

rjelbert, 2019-01-09:
Thank you. I'll give it a go!

StefanScherer, 2019-09-16:
Closing due to inactivity

closed 2019-09-16 by StefanScherer