hypriot/image-builder-rpi #278: Makefile:7: recipe for target 'sd-image' failed

AnsgarSchmidt, 2019-01-31:
Hi, 

I try to build an image with the current git version. It worked a week ago but now it seems there is a key expired.

gpg: directory '/root/.gnupg' created
gpg: keybox '/root/.gnupg/pubring.kbx' created
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
gpg: /root/.gnupg/trustdb.gpg: trustdb created
+ FINGERPRINT_OF_FILE=6A037BB52DF7D46D99DC59C101666247EBFF1218
+ [[ 40 -eq 16 ]]
+ [[ 40 -eq 8 ]]
+ echo 'compare fingerprint'
compare fingerprint
++ clean_print fpr 6A037BB52DF7D46D99DC59C101666247EBFF1218
++ local fingerprint=6A037BB52DF7D46D99DC59C101666247EBFF1218
++ local func=fpr
++ nospaces=6A037BB52DF7D46D99DC59C101666247EBFF1218
++ tolowercase=6a037bb52df7d46d99dc59c101666247ebff1218
++ KEYID_long=01666247ebff1218
++ KEYID_short=ebff1218
++ [[ fpr == \f\p\r ]]
++ echo 6a037bb52df7d46d99dc59c101666247ebff1218
+ CHECK=6a037bb52df7d46d99dc59c101666247ebff1218
+ [[ 418a7f2fb0e1e6e7eabf6fe8c2e73424d59097ab == \6\a\0\3\7\b\b\5\2\d\f\7\d\4\6\d\9\9\d\c\5\9\c\1\0\1\6\6\6\2\4\7\e\b\f\f\1\2\1\8 ]]
+ echo 'key invalid'
key invalid
+ exit 1
Makefile:7: recipe for target 'sd-image' failed
make: *** [sd-image] Error 1
mv: cannot stat 'hypriotos-rpi-dirty.img.zip': No such file or directory
mv: cannot stat 'hypriotos-rpi-dirty.img.zip.sha256': No such file or directory

AnsgarSchmidt, 2019-02-02:
On a second thought the problem seems to be the way the key is extracted 
[[ 418a7f2fb0e1e6e7eabf6fe8c2e73424d59097ab == \6\a\0\3\7\b\b\5\2\d\f\7\d\4\6\d\9\9\d\c\5\9\c\1\0\1\6\6\6\2\4\7\e\b\f\f\1\2\1\8 ]]

kraeml, 2019-02-07:
A dirty workaround

I changed in builder/chroot-script.sh 

```
PACKAGECLOUD_FPR=6A037BB52DF7D46D99DC59C101666247EBFF1218
```

May the key is wrong or old?

I ran on old hypriot pis this

```
gpg_key_url="https://packagecloud.io/Hypriot/rpi/gpgkey"
curl -L "${gpg_key_url}" 2> /dev/null | apt-key add - &>/dev/null
apt update
```

Because the apt-key was broken.

This was found on:
https://packagecloud.io/Hypriot/rpi/install
or better
https://packagecloud.io/install/repositories/Hypriot/rpi/script.deb.sh