hypriot/image-builder-rpi #291: Permanent arm64 build

neerdoc, 2019-04-30:
Hi all!

I really like this project and have built a 10-unit cluster of Raspberry Pi 3B+ units to run kubernetes. However, I would really like to use _rook.io_ as storage solution which fails due to 64-bit requirements.

I found the link in the blog to the [64-bit fork](https://github.com/DieterReuter/image-builder-rpi64/releases), but it seems a bit stale. Thus, I am wondering if it would be possible to actually build this image for both 64-bit and 32-bit arm here?

I have a fork of my own of this repo and have tried to make a generalized build where you can choose 32-bit or 64-bit (based on the @DieterReuter 's build), but I currently stuck after running the _cloud-init_ script during booting. 

So a couple of things:
1. Discussion: Is a dual-architecture build version a good thing to have here?
2. Should I submit a PR on my _not yet working_ attempt?
3. Any suggestions on why [my build](https://github.com/neerdoc/image-builder-rpi/tree/build-arm64) is stuck?

DieterReuter, 2019-05-01:
I really like the idea to have both 32bit and 64bit builds in one place. But I'm not sure if this is easy to do, because in 64bit repo there are quite some differences to the 32bit build process. At least we should make sure that both builds are in sync.

neerdoc, 2019-05-05:
I believe most differences are because of the 32-bit build is based on _raspbian_ whereas the 64-bit build is based on _debian_. Maybe the differences would be small if both where based directly on _debian_?

sotiris84, 2019-08-10:
I would love to see a true 64bit version of hypriot. Hopefully this gets picked up by the devs.

foscoj, 2019-12-06:
Maybe [this ](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=250730) development could enable the raspbian (and therefore platform optimized) based 64bit hypriotOS @DieterReuter ?
If it is focused on rpi4 first and backported later, depending on kernel development, this might be the way to go in my opinion.

Zakspeed, 2019-12-10:
I would see a true 64bit version of hypriot, 

https://ubuntu.com/blog/updated-images-of-ubuntu-for-the-raspberry-pi-2-3-and-4

Ubuntu Server 19.10.1 ARM64
or: 
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=250730
Force Raspbiam 64bit

616b2f, 2020-01-29:
@neerdoc to fix your build error, this PR could be helpful https://github.com/DieterReuter/image-builder-rpi64/pull/116

616b2f, 2020-02-20:
@neerdoc  I had time to look into your build issue, it fails on:

```sh
+ '[' '!' -f /workspace/arm64/rootfs-arm64-raspbian-v2.0.1.tar.gz ']'
+ wget -q -O /workspace/arm64/rootfs-arm64-raspbian-v2.0.1.tar.gz https://github.com/hypriot/os-rootfs/releases/download/v2.0.1/rootfs-arm64-raspbian-v2.0.1.tar.gz
make: *** [sd-image] Error 8
Exited with code 2
``` 

the thing is there are no raspbian package in releases for arm64, only this:

rootfs-arm64-debian-v2.0.1.tar.gz
rootfs-armhf-debian-v2.0.1.tar.gz
rootfs-armhf-raspbian-v2.0.1.tar.gz

maybe we could try debian instead or create arm64 on raspbian.

Can someone say how hard it would be to build arm64 raspbian package.

Zakspeed, 2020-02-21:
https://ubuntu.com/download/raspberry-pi

Download "64-bit  Ubuntu 18.04.4 LTS"

StefanScherer, 2020-02-21:
Thanks @Zakspeed for the heads-up.

geerlingguy, 2020-06-14:
Note the new [Raspberry Pi OS](https://www.raspberrypi.org/downloads/raspberry-pi-os/) currently has a [64-bit OS beta](https://www.raspberrypi.org/forums/viewtopic.php?t=272204) that looks like it's even more directly built from Debian Buster.

Not sure where that puts things for HypriotOS, but it seems likely that maybe both OSes will converge to be somewhat closer to Debian at some point (and less Raspbian-ish)?

Legion2, 2020-07-18:
I tried the stable 64 bit kernel by setting `arm_64bit=1` in `/boot/config.txt` and it just works. The only problem is, that the whole distro is still 32 bit so I can't even run 64 bit docker container as docker still think it's 32 bit.

I need 64 bit support for influxdb which have some OOM problems with the 32 bit version.