hypriot/image-builder-rpi #292: v1.10.0 beta3 should use docker test repo

deftdawg, 2019-05-04:
Updating the system with ``sudo apt update; sudo apt-upgrade`` downgrades docker to the latest stable release 18.09 which then coredumps...

Downgrade can be prevented by editing ``/etc/apt/sources.list.d/docker.list`` and changing ``stable`` to ``test`` before running apt update/upgrade.

StefanScherer, 2019-05-04:
Nice catch, thanks @deftdawg 
Do you want to make a PR to fix that?

deftdawg, 2019-05-04:
I don't need the credit, not even sure that's is the best way to solve it.

If the goal is stable releases of hypriot, some apt pinning policy that
achieves the equivalent of ``dpkg --refuse-downgrade`` might be a better
overall approach.  I.e. hold the installed version until something in
stable gets a higher version.

deftdawg, 2019-05-05:
Best approach I could come up to solving this was to make a dummy package that depends on >= installed version of docker-ce...  This should allow the system to install a newer version when it appears in "stable", but prevent "stable" from downgrading below what ships installed on the image.

Not sure the best place to hook it in, but ideally it would on the first boot as either root or another user that can sudo without a password.

generate-docker-dummy-package.sh:
```bash
#!/usr/bin/env bash
# Deftdawg's quick and dirty script to generate dependency dummy package
# Use this on first run to prevent docker from being downgraded...
PACKAGE=docker-ce_upgrade-only
VERSION=$(date +%Y-%m-%d.%H.%M)

pushd /tmp
(cat << EOF
Package: ${PACKAGE}
Version: ${VERSION}
Architecture: all
Maintainer: Deft Dawg <deftdawg@gmail.com>
Installed-Size: 32
Depends: docker-ce (>= $(dpkg -s docker-ce | grep -e "^Version:" | cut -d" " -f2-))
Section: admin
Priority: optional
Homepage: https://github.com/hypriot/image-builder-rpi
Description: Dummy dependency package to prevent docker-ce from
 downgrading by depending on docker-ce >= image version shipped,
 it will not allow downgrade but will allow upgrade when "stable"
 repo version reaches a higher revision.
EOF
) > control
touch md5sums
tar -zcf control.tar.gz control md5sums
tar --exclude=* -cvf - . | xz - > data.tar.xz
echo "2.0" > debian-binary

rm ${PACKAGE}_*.deb # remove any older copies of this deb
ar qf ${PACKAGE}_${VERSION}.deb debian-binary control.tar.gz data.tar.xz
rm control control.tar.gz debian-binary md5sums data.tar.xz

sudo dpkg -i ${PACKAGE}_${VERSION}.deb
popd
```

deftdawg, 2019-05-06:
Just did a fresh install of v1.10.0, this must've been what happened before:

```
apt install --dry-run docker-ce
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  containerd.io docker-ce-cli
Recommended packages:
  aufs-tools cgroupfs-mount | cgroup-lite pigz
The following NEW packages will be installed:
  containerd.io docker-ce-cli
The following packages will be upgraded:
  docker-ce
1 upgraded, 2 newly installed, 0 to remove and 24 not upgraded.
Inst containerd.io (1.2.5-1 Docker CE:stretch [armhf])
Inst docker-ce [18.06.3~ce~3-0~raspbian] (5:18.09.0~3-0~raspbian-stretch Docker CE:stretch [armhf]) []
Inst docker-ce-cli (5:18.09.0~3-0~raspbian-stretch Docker CE:stretch [armhf])
Conf containerd.io (1.2.5-1 Docker CE:stretch [armhf])
Conf docker-ce (5:18.09.0~3-0~raspbian-stretch Docker CE:stretch [armhf])
Conf docker-ce-cli (5:18.09.0~3-0~raspbian-stretch Docker CE:stretch [armhf])
```
The result was docker core dumping, since the docker releases are the same for v1.10.0, it may happen again unless whatever was causing that behaviour was fixed for 1.11.0 (maybe something kernel or something).

If it happens again, I will do:
```yaml
runcmd: 
   - 'apt-mark hold docker-ce'
```

If not I will close the issue, as it will be fixed with 1.11.0

closed 2019-05-06 by deftdawg