hypriot/image-builder-rpi #295: Docker service takes long to start

Duvel, 2019-06-08:
I found out that with version 1.10 installed on some devices it could take docker several minutes before it started. I was researching another issue with a service that wouldn't start at boot and found out that docker wasn't starting quickly. Checking the logs didn't help and adding debug didn't give any solution as well.

Then I went to check out the 1.11-rc2 and found that it was starting quickly and I tried to install that docker version on an 1.10 build.

The new docker version did give me a hint, why it wasn't starting:

```crypto/rand: blocked for 60 seconds waiting to read random data from the kernel```

Then I installed haveged, which sometimes is also needed for hostapd, and that seemed to solve my problem and docker start immediately.

StefanScherer, 2019-06-08:
So, it doesn‘t take too long with 1.11.0-rc2?

Duvel, 2019-06-08:
For now the 1.11.0-rc2 looks good, but I haven't tested it yet on a Raspberry Pi 0 W, where I first found the problems. In fact I'm not using docker yet on these machines, so that could be part of the problem.

Duvel, 2019-06-27:
Found the time to do some testing. It also occurs with the 1.11.0-rc2.

It happens probably because I remove cloud-init, as I take over with ansible after the first boot.

Duvel, 2019-06-27:
Confirmed it now on a Raspberry Pi 1, only removing cloud-init makes docker wait for the system to get enough randomness.

closed 2019-09-16 by StefanScherer