hypriot/image-builder-rpi #296: apt update/upgrade in cloud init not working

cranphin, 2019-06-15:
Hi!

I love your images, currently using 1.10.0.
Note: Just noticed 1.10.1 is out too, just not linked from https://blog.hypriot.com/downloads/, I'll see if that fixes the issue :)
I'm trying to install on a new Raspberry Pi 3 b+
I flash the image to an sdcard, and put a user-data file in the root of the fat partition.
The I just boot the card in the Pi and wait.

I would have expected all the apt packages to be updated.
Instead I get the following error in cloud-init-output.log:

```
WARNING: The following packages cannot be authenticated!
containerd.io docker-ce docker-ce-cli
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
```

I'm not sure how to fix this, I tried 'curl -fsSL https://download.docker.com/linux/ubuntu/gpg | apt-key add -', but that didn't help.

My user-data is as follows:
```#cloud-config
# vim: syntax=yaml

hostname: black-pearl
manage_etc_hosts: true

users:
  - name: pirate
    gecos: "Hypriot Pirate"
    sudo: ALL=(ALL) NOPASSWD:ALL
    shell: /bin/bash
    groups: users,docker,video,input
    plain_text_passwd: blabla
    lock_passwd: false
    ssh_pwauth: true
    chpasswd: { expire: false }
    ssh-authorized-keys:
      - ssh-rsa BLABLA

timezone: "Europe/Amsterdam"

package_update: true
package_upgrade: true
package_reboot_if_required: true

packages:
  - screen

write_files:
  - content: |
      allow-hotplug wlan0
      iface wlan0 inet dhcp
      wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
      iface default inet dhcp
    path: /etc/network/interfaces.d/wlan0
  - content: |
      country=nl
      ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
      update_config=1
      network={
      ssid="BlaBla"
      psk="blabla"
      proto=RSN
      key_mgmt=WPA-PSK
      pairwise=CCMP
      auth_alg=OPEN
      }
    path: /etc/wpa_supplicant/wpa_supplicant.conf

runcmd:
  - 'systemctl restart avahi-daemon'
  - 'ifup wlan0'
  - 'sleep 10'
  - 'curl -fsSL https://download.docker.com/linux/ubuntu/gpg | apt-key add -'
```

All the apt output from cloud-init-output.log:
```
Cloud-init v. 0.7.9 running 'modules:config' at Sun, 24 Feb 2019 17:32:57 +0000. Up 31.50 seconds.
Get:1 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Ign:4 https://download.docker.com/linux/raspbian stretch InRelease
Get:5 http://archive.raspberrypi.org/debian stretch/main armhf Packages [221 kB]
Ign:6 https://download.docker.com/linux/raspbian stretch Release
Ign:7 https://download.docker.com/linux/raspbian stretch/stable all Packages
Ign:8 https://download.docker.com/linux/raspbian stretch/stable armhf Packages
Ign:9 https://download.docker.com/linux/raspbian stretch/stable Translation-en
Get:10 http://raspbian.raspberrypi.org/raspbian stretch/contrib armhf Packages [56.9 kB]
Get:11 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [95.5 kB]
Get:12 http://raspbian.raspberrypi.org/raspbian stretch/rpi armhf Packages [1,360 B]
Ign:13 https://download.docker.com/linux/raspbian stretch/stable Translation-en_US
Ign:7 https://download.docker.com/linux/raspbian stretch/stable all Packages
Ign:8 https://download.docker.com/linux/raspbian stretch/stable armhf Packages
Ign:9 https://download.docker.com/linux/raspbian stretch/stable Translation-en
Ign:13 https://download.docker.com/linux/raspbian stretch/stable Translation-en_US
Ign:7 https://download.docker.com/linux/raspbian stretch/stable all Packages
Ign:8 https://download.docker.com/linux/raspbian stretch/stable armhf Packages
Ign:9 https://download.docker.com/linux/raspbian stretch/stable Translation-en
Ign:13 https://download.docker.com/linux/raspbian stretch/stable Translation-en_US
Ign:7 https://download.docker.com/linux/raspbian stretch/stable all Packages
Get:8 https://download.docker.com/linux/raspbian stretch/stable armhf Packages [5,039 B]
Ign:9 https://download.docker.com/linux/raspbian stretch/stable Translation-en
Ign:13 https://download.docker.com/linux/raspbian stretch/stable Translation-en_US
Ign:7 https://download.docker.com/linux/raspbian stretch/stable all Packages
Ign:9 https://download.docker.com/linux/raspbian stretch/stable Translation-en
Ign:13 https://download.docker.com/linux/raspbian stretch/stable Translation-en_US
Ign:7 https://download.docker.com/linux/raspbian stretch/stable all Packages
Ign:9 https://download.docker.com/linux/raspbian stretch/stable Translation-en
Ign:13 https://download.docker.com/linux/raspbian stretch/stable Translation-en_US
Fetched 12.1 MB in 110d 6h 37min 1s (1 B/s)
Reading package lists...
W: The repository 'https://download.docker.com/linux/raspbian stretch Release' does not have a Release file.
Reading package lists...
Building dependency tree...
Reading state information...
Calculating upgrade...
The following NEW packages will be installed:
containerd.io docker-ce-cli
The following packages will be upgraded:
base-files dbus docker-ce libasound2 libasound2-data libdbus-1-3
libpam-systemd libpng16-16 libraspberrypi-bin libraspberrypi0 libssh2-1
libssl1.0.2 libsystemd0 libudev1 openssh-client openssh-server
openssh-sftp-server python3-cryptography raspberrypi-bootloader
raspberrypi-kernel systemd systemd-sysv tzdata udev wget wireless-regdb
wpasupplicant
27 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 78.7 MB of archives.
After this operation, 17.0 MB of additional disk space will be used.
WARNING: The following packages cannot be authenticated!
containerd.io docker-ce docker-ce-cli
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
Cloud-init v. 0.7.9 running 'modules:final' at Sun, 24 Feb 2019 17:33:06 +0000. Up 40.50 seconds.
2019-06-15 02:10:22,866 - util.py[WARNING]: Package upgrade failed
```

muelli, 2019-08-09:
That looks like https://github.com/hypriot/image-builder-rpi/issues/304

Legion2, 2020-01-12:
This should now be fixed with hypriot/flash#162 in the latest release of the image and flash tool. The problem was, as you can see in the logs that the time was not correct (`at Sun, 24 Feb 2019`). So the package signatures were not valid at that time. The problem was solved temporarily with the new release, because the time of the new release was not too far in the past when you tested it.

eshkrab, 2020-08-07:
I'm also still getting the problem. running flash version 2.7.0