hypriot/image-builder-rpi #298: Allow to optionally replace docker by k3s

guidoffm, 2019-06-21:
Maybe related to #249 and #208.

Rancher published a lightweight Kubernetes implementation named K3s that does not use Docker. To create a Kubernetes cluster on RPi this would be interesting.

jeefberkey, 2019-06-21:
I've done this, it's very very easy. No changes to this image are required, just go install k3s. You can remove docker if you want.

StefanScherer, 2019-09-08:
We have no plans to do that. It would mean that we have to create another SD image.

closed 2019-09-08 by StefanScherer

pbolduc, 2021-02-01:
> I've done this, it's very very easy. No changes to this image are required, just go install k3s. You can remove docker if you want.

I am trying to set this up locally. I am new to this. I think the simplest way is to disable docker at start up? I know you say it is very very easy, but I would appreciate if someone could provide a little detail. I would think I could adjust docker start up using `systemd` in the cloud init or create an ansible role to disable.  I just need a nudge in the right direction.