hypriot/image-builder-rpi #299: Raspberry Pi 4 incompatibility with HypriotOS v1.11.0-rc2

rjelbert, 2019-06-25:
I am trying to get HypriotOS v1.11.0-rc2 to boot on a Raspberry Pi 4. It won't boot. I swapped to the new Buster Raspbian SD card and it booted so hardware OK. I don't have the micro HDMI adapter yet so I can't include the specific error message but I will update tomorrow once received.
I already re-flashed the boot loader as recommended by Raspberry Pi .org

phillebaba, 2019-06-27:
I got my Raspberry Pi 4 today, so I could also test. I don't think I have enough knowledge about how the OS is built though, so I have little to offer code wise.

StefanScherer, 2019-06-27:
I also got my RPi 4 today, but I'm not at home 😢

d11wtq, 2019-06-29:
Confirmed it doesn't boot on the Raspberry Pi 4. I have the 4GB one and when powered on with Hypriot the green LED blinks 2-3 times and then goes off and all the remains is the power LED. Raspbian works fine. Assume something has changed with the required boot configuration, but not much of an expert to know what.

d11wtq, 2019-06-29:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

d11wtq, 2019-06-29:
Update: I was able to get it to boot and was clearly operating because the disk activity LED was regularly flashing away like a normally functioning Pi would do. However I couldn't connect to the Pi over ethernet, presumably because of the documented network driver change mentioned in the documentation above.

Full disclaimer, I have no idea what I'm doing. This is what I did:

  1. Took the latest Raspbian release (which I know to work fine) and copied its /boot file system verbatim to my desktop machine.
  2. Took the latest stable (1.10.1) HypriotOS release and did the same thing.
  3. Flashed the SD card with HypriotOS 1.10.1.
  4. Overwrote the entire /boot on the flashed SD card with the one I copied from Raspbian.
  5. Copied kernel.* and cmdline.txt over from the original HypriotOS /boot back over the top of the Raspbian /boot directory (so it was using the same HypriotOS kernel, in theory). Only issue here is there is a kernel7l.img on the Rasbpian /boot that doesn't exist on Hypriot (like I say, I have no idea what I'm doing).
  6. Inserted the SD card and booted the Pi. Blinking green light (non regular) exactly like a functioning Pi. This continues for many minutes. I have since powered off the Pi to continue experimenting.

DieterReuter, 2019-06-29:
It seem you now lack the ethernet driver files which should be part of the root filesystem, these are of course currently not included in HypriotOS.

But great work! Thanks for sharing these details.

webbe035, 2019-07-16:
@d11wtq I was able to get a version of HypriotOS to run on my PI4 using your instructions but with some minor tweeks:
-You need to copy the missing .dtb files in the overlay folder from Raspbian to the HypriotOS image. (there are around 25 missing)
-On startup I encountered issues with the cloud-init procedure requiring a setup of a new user account (standard pirate and hypriot credentials didn't work for some reason)

StefanScherer, 2019-07-18:
See work in progress #300.

webbe035, 2019-07-19:
I just flashed an image from the new branch and was able to get all my containers to run, thanks a lot!

rjelbert, 2019-07-19:
Can you point me to the SD card image download so I can test on our Pi4's? Thanks

StefanScherer, 2019-07-19:
@rjelbert it's in PR #300, follow the green arrow to circle, then you have the builds for this branch, under Artifacts there is always a img.zip file, eg. in https://circleci.com/gh/hypriot/image-builder-rpi/183#artifacts/containers/0
But it's too early for tests on RPi4 right now, it didn't work for me today.
work in progress:
- make cloud-init 18.3 work
- investigate what's still missing for RPi4

Duvel, 2019-07-20:
I just gave the image from #300 a spin on my Raspberry Pi 4 B with 2 GB and it worked, both WiFi and cabled. I was able to login. If you need some logs or so, just say so.

Duvel, 2019-07-20:
After some time spent, I also found a way to get cloud-init working again.

I added these lines in /etc/cloud/cloud.cfg: 
```
datasource_list: [ NoCloud ]

datasource:
  NoCloud:
```

And I added this in /boot/cmdline.txt:
```
 serial=ds=nocloud;s=/var/lib/cloud/seed/nocloud-net
```

I think the files now should be symlinked in /var/lib/cloud/seed/nocloud, as nocloud-net allows for a download from a webserver if I understand correctly.

StefanScherer, 2019-07-20:
Thanks for testing @Duvel
I‘ll prepare a new sd image with the cmdline.txt later this weekend.

StefanScherer, 2019-07-21:
Fixed with #300. Check out the pre-release v1.11.0-rc3

closed 2019-07-21 by StefanScherer