hypriot/image-builder-rpi #301: Raspberry Pi 3+ wifi setup

theTranqu, 2019-07-18:
Hi,
i want to seed multiple raspberrys to form a cluster and am following the instructions for flashing https://github.com/hypriot/flash and in general this one for setting up the cluster: https://blog.hypriot.com/post/setup-kubernetes-raspberry-pi-cluster/

Yet I can not get the first raspberry online- i looked around a bit and it seems the current rc2(https://github.com/hypriot/image-builder-rpi/releases/tag/v1.11.0-rc2) does not actually have working wifi, according to this: (https://github.com/DieterReuter/image-builder-rpi64/releases) 

> WiFi works now on Raspberry Pi 3 Model B+ (kudos to @ttyS0)

I tried  both the hypriot latest release aswell as the latest one from @DieterReuter , yet I had no joy yet. 
What is actually still required for wifi? The no-uart- config file, is it actually still needed at this time?

intershopper, 2019-07-21:
I have faced ecxactly the same issue. Have wlan0 interface is down and there is no connection or ip address from from access point. 
Regards, 
Frank

StefanScherer, 2019-07-21:
I just tried a RPi 4 with this wifi.yml

```yaml
#cloud-config
# vim: syntax=yaml
#

# Set your hostname here, the manage_etc_hosts will update the hosts file entries as well
hostname: aaabbb
manage_etc_hosts: true
# don't write debian.org into apt mirrors
apt_preserve_sources_list: true

# You could modify this for your own user information
users:
  - name: stefan             # use any user name you like
    primary-group: users
    shell: /bin/bash
    sudo: ALL=(ALL) NOPASSWD:ALL
    groups: users,docker,adm,dialout,audiolugdev,netdev,video
    ssh-import-id: None
    lock_passwd: true
    ssh-authorized-keys:
      - ssh-rsa AAAAxxxxxx

# # Set the locale of the system
# locale: "en_US.UTF-8"

# # Set the timezone
# # Value of 'timezone' must exist in /usr/share/zoneinfo
# timezone: "America/Los_Angeles"

# # Update apt packages on first boot
# package_update: true
# package_upgrade: true
# package_reboot_if_required: true
package_upgrade: false

# # Install any additional apt packages you need here
# packages:
#  - ntp

# # WiFi connect to HotSpot
# # - use `wpa_passphrase SSID PASSWORD` to encrypt the psk
write_files:
  - content: |
      allow-hotplug wlan0
      iface wlan0 inet dhcp
      wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
      iface default inet dhcp
    path: /etc/network/interfaces.d/wlan0
  - content: |
      country=de
      ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
      update_config=1
      network={
      ssid="aaabbbccc"
      psk="xxxyyyzzz"
      proto=RSN
      key_mgmt=WPA-PSK
      pairwise=CCMP
      auth_alg=OPEN
      }
    path: /etc/wpa_supplicant/wpa_supplicant.conf

# These commands will be ran once on first boot only
runcmd:
  # Pickup the hostname changes
  - 'systemctl restart avahi-daemon'

  # Activate WiFi interface
  - 'ifup wlan0'
```

and 

```
flash --userdata wifi.yml -n blue hypriotos-rpi-dirty.img
```
and I was able to just SSH into it with `ssh blue.local`

StefanScherer, 2019-09-16:
Closing due to inactivity.
Did multiple flashes with 1.11.1 and 1.11.2 today with Pi Zero W and Pi 4 with WiFi enabled and it worked. Adjust ssid and psk in the sample above.

closed 2019-09-16 by StefanScherer