hypriot/image-builder-rpi #305: ERROR: for [name]  Cannot start service [service]: OCI runtime start failed: container process is already dead: unknown

paulomcnally, 2019-07-25:
```
hypriotos-rpi-v1.10.0.img.zip
```

```
$ docker -v
Docker version 18.06.3-ce, build d7080c1
```

Error message:
```
ERROR: for [name]  Cannot start service [service]: OCI runtime start failed: container process is already dead: unknown
```

docker-compose.yml
```
version: '3'

services:
  postgres:
    image: postgres:11.2
    ports:
      - 5432:5432
    environment:
      POSTGRES_PASSWORD:
  redis:
    image: redis:5.0.3
    ports:
      - 6379:6379
    command: redis-server --appendonly yes
  mongo:
    image: amd64/mongo:4.1-xenial
    ports:
      - 27017:27017
```

## Solutions

https://success.docker.com/article/error-oci-runtime-create-failed-docker-1809

StefanScherer, 2019-07-25:
At least this image `amd64/mongo:4.1-xenial` is not available for the Raspberry Pi CPU, also the `mongo:4.1` only provides an amd64 and arm64 image.

StefanScherer, 2019-09-16:
Closing due to inactivity

closed 2019-09-16 by StefanScherer