hypriot/image-builder-rpi #307: cloud-init apt installs original debian sources without key

muelli, 2019-07-30:
I have tried to add an additional sources.list.d entry through cloud-init. The entry was correctly placed, but along with that came debian sources which were not there before. I think https://github.com/hypriot/image-builder-rpi/blob/34b9e1b18dfc4371bf37db3bbae4320bcc5dc1fe/builder/files/etc/cloud/cloud.cfg#L99-L103 is the culprit.

the problem manifests itself in not being able to do an apt-get update because the keys seem to be missing. Also, I cannot installed packages, but I have not understood why.

My user-data has this:
```
apt:
    sources:
        comitup.list:
            source: "deb http://davesteele.github.io/comitup/repo comitup main"
            key: |
                -----BEGIN PGP PUBLIC KEY BLOCK-----

                mQINBExnT3cBEACjM5fQ6hQUW6Rp7ddUzEhW3VKZ6cub3kdXMINk0Zrbyw9+J4T9
                SKeDe9h0OpRr2GtbckMT9tbyTfU7lB62uUZQpd/aPVghQtS+ztcWcDiCCN4+Yq6J
                FNLWJWd3VbRB5EmBYxC1/itGFn2TTMnYQPTT9z/fwZA634rPUonGurKS3mwdtWHK
                0vhuyWu4BtiMfCsrO3T+SiPAKgmmUQAMW5ySwxPVlZVgdC5r4X7gwZyk3sw4gtt7
                hN6a/9MIA5N56Ns5UD5yyO1NFph2P6OVqB8fca2aKIwiXrqKhZp4MvT/hZA2F3Sj
                i355/d44durtJInFztO25T4DlUkAsLAMhewtYBuS0k7fHQF2eskj1Z3J74e2b7ae
                F05FBG5Gfi9R+XMNFFFtjivlvdrkgTEgaoHue/VUuvvyd+eWrAg3yfM87qY/kP0H
                ZJlo4dEa9p2lqa4fhRDMk4PgzPWohIpjKyztG3QGSO/ix+wvE6avalB+FMqCcs0/
                vkMfHIARW/4PnGKHFhfb+uDW1CyO9TC4uOTAMbFWQKH+GQ+YxKHBDkbVhXnZRILq
                oiXf56m7QYOwnPZ63/4WW5y3mIscEKzx+0HUDG5u9ds7TTrZhUlUQR1bmnoiFIYq
                5uA20SGu/v+VJ8Yg7qticOl1J6cH9NugXq0Fy7yHCzLav4GOAMuo8Q2RKQARAQAB
                tCBEYXZpZCBTdGVlbGUgPHN0ZWVsZUBkZWJpYW4ub3JnPokCNwQTAQgAIQUCV+J8
                fwIbAwULCQgHAwUVCgkICwUWAgMBAAIeAQIXgAAKCRCKMXHvNmFQzpUvEACeI2r2
                2KVxLEoMGcHHguZBZqr4BGsKvMxPcz9eLlNknRCnh0x5fmXE2128YvsA5xJva6B4
                hYFw9gOw7WSjgdgmdVj6zTECRd6ERDp2GHtNKsW5e7FpsLQqAubxWpc0/zKs8Vaf
                RQ7C8kL5zxDa7w3KyeBgM0FPcr6KTYsNd8okJmvDKXht0YlH75VLTbGCJqVMRrMm
                JCNlADdJD2LddThtwTiPhC6JlW+JI0j5bwZ4L6QQARXxZk8Chw/7s0DY2lEOAcdJ
                dTjAR2VLbc20AYeZ0yu9NkgrJrCweRklADemgvmPEf6ls7UpTPb7vzi3MKqfMtBh
                UHtOUb9j9wzI+wA9glgajE0PEqrm3JvWwTdj9dpWLWQgzWUzwrs+5qquE3xPZsBD
                tjgaLxP/Dqd4hE2agef4BFGQHAVnzju70GpdJWvGGrR+s+k7F9mNUkg7WB9tegj5
                3oT1mu4Ciq7qUa3X0n73iQawwfmLTl1igC1kZr3ab7GezobUcswAdn692ufUx4Vt
                54o2kOC6fZEWfVEmU+uTcEk29GvdoAl04EdSg9Hrof8A/Z4yKvxP3QtXHOVLY5gg
                cZ0KscZcD6+AvmMyQ6cxbHOl62T5O7HUy63K+8ELCt9PSZJuTihhbgvVi4TJWfDk
                UJOz9UdZc4HpS6862BIZtoVfQuoNat35RXHg4bQqRGF2aWQgU3RlZWxlIDxkYXZl
                c0B1c2Vycy5zb3VyY2Vmb3JnZS5uZXQ+iQJEBDABCAAuFiEErg2/WpKlreSUgbpv
                ijFx7zZhUM4FAllPIZsQHQBPYnNvbGV0ZSBlbWFpbAAKCRCKMXHvNmFQzt4DD/9l
                m+Be+FO6D2cgKjw0cYPgXxoVodUzJG+GcXMafagFREM4HD6JmYsoZN/VlnFFWHwf
                fOmnFg9cZ/W4nQAv/kTF6a3w31lDUmjV45/9AiGAMxL+oodQpGt2cOBkB5T/5clJ
                w3PcEKPNgADYZTzzgxC8E+Rm63iIFfAfoA7noiBBFuictGhGeKbuPy0zqS9I+oq2
                YSvHP7yMV4RsU2apvQRyU8uFcXbwRVpL4t2EC/4eWM/sG7EuscKbVPjUcQLTPMUE
                7Z6mAFp2AtZqPKRWCc8E8I9FCpmp4dQSyMIRdhQHMpcsij90w69CEr2v/KyFeO4N
                gG00YbA3oGGxYZYoPnsRkDAk5MxAspMxtBUNil4uJaLvNBolXtHIj2YWDtTF5+dp
                PVCUC8pJem1wKV9TFfprLLmh3s+nythw7wRY96IAvskZlpE61EuWhr57FeRuippj
                /+GJeqWpUmXPG3S1dm9KGujCI/o2IOKhug3qqupE4T5lzjdkJkLHmzrHZ1N7BQEh
                moxbcSFS3iJi9LzJPWLI7TOipKmg5QUkZPFpqtASIvzRltpfdi75EXQHt6OBbTzv
                kkkuh5NA0YQ/ambvjfKOkLRrmyYE0ryH+0l8py+os3nq/0r0MZG/SbxOSGHxHZJs
                QkQsh6CCBTm6cFJsTkLbgW/mfRssJanCOEPoegzKfLQgRGF2aWQgU3RlZWxlIDxk
                c3RlZWxlQGdtYWlsLmNvbT6JAjcEEwEIACEFAlQrY70CGwMFCwkIBwMFFQoJCAsF
                FgIDAQACHgECF4AACgkQijFx7zZhUM6tAA//RIC2cIHt9npRg+RJy8z9ClWSyeHv
                qNxML7WflnfCBKpOwwMAC8FLLHwFonf4cvK7uKkolBvOTAoyjmNicLA+zHE7T7y+
                Vg7bhkd/KVNTjVn3qqkXs0GDCod+ztzXKlJG9Z3rD/XeFN8OrHZW3CZLgNkIga5W
                qamX2a404/ZMn346pj2TJM1jWue7WALUn7846FM08+rzo7rKLWwKmYUpWj2YRm37
                Sj34ldv+sIc5SDCUsDDnGDxtB2evCYiy09tm9gFxNgtMdGe5S/Igsd/eEpg1ARoT
                tV4sFZK/3BDxjWFKqE0dGRp1UZxPPRExumGUe+oW6kyzJnO21m26p+8/T9ggYq21
                u2POkyoD9Q7YerOM/3h1d5W51Hv6fzS5uXASJ+yua9BPwB20NBnayiB6G50poxXo
                HGpk4BqDbC7q7Pmh69U+a/fWtEFj9QFQdUIbXV/fVEK8GpOCEmHWDiXrRHAdai8r
                6mjZnisfn3ipc0fqCTjR7XEFAORz6At/59zgU7YeQsAkrxuA9L1X9DEbeHgiE51x
                cYZVQdrBDxW2q3jtv7tyS50EiEE9TQFEMLHA5rxfpZLFluO8ZEtN7l6f8kPbsW2z
                sb1aF0+jLb0A7gYKszyyKNkjcryvmMQCn2kYjxDcLFFXDgsncM86pLNv+lML6G6b
                ZbygTrc00n+cbR+5Ag0ETGdQCAEQANYUWyxyHeKpvB2VUjMZUvHqggplYA6ouguy
                MrZ4/cPvXFEWxSz/63d2fUj2nz6Uhg6RnvinEiXEcLG1YG+zegSQm2DfBY5VRa3G
                nqv7+N46vn+WZzlRhVCRUT2qBQgCeqNfzhLz6m2sn7Z6XBF7PeaN/dLsORketbFf
                fIx7P7hVqmdOUh4RIeSgvmgsWdJRzCfY8a0roORv8UOwWBYW5OCKed+VMBftKMoO
                w4zstSoN0b98wULq5B11wtxH1fVuuxpb+Zw7Ab+p9+AMXhaPnrRCvAOS8ou1Zdh2
                LE/p/M9JEolOUOo/66qpL5ix2Y/+g3YaNDQqmNMHTy/QlbTIdw/nn3yLO27etGnE
                9TMcli10rLYGiuOzy4sU6K3RzGGJyQpif3FAsvkIPXtFlBqd4cdN2dfJ6hqUtnbP
                u4/yfm8M+f/UYrOEAL52R602JEWWu3pH2j3PZhfsPKwTMd14jATJHxJTFyc6odp0
[... 63 lines cut ...]
                2wI5VLqBJk8KhhYHZp2aVs0hlluLvXu4j5ocDKV0kZNyFP5Dbr9JaRmbFWT45tMu
                ZQcOx0IBFVQee66s9swo8B/m/75fktMGTqLXyBJrBgIOQnJd1+sT2GYF3fN5RSaA
                xsPfrdnKDV9zaFqUyPkLYHdn30SJOLR7xUngaBtDN7K+Br+WQljm03q4w/EoTonB
                FCpYE8uADKi2XHSXujwsHTi9GCuZd8EmXufv3pKdvQ0op/8XoEQ+lkksHg0OQTuO
                mhq49uPU+W1hJ9y7OtpzwYdIvlU06Y1XFV8Se0uIhvCIyHsK9ox3F6152vWJckEy
                8PiUoTVmQk6ilTck24F9nPk2DTfur4GKm0rHk6hhhjVFCufToDe/g6ANSEUHZVyM
                lGNF2YVGW29TPMejWzv9ysMz3QCWFNe8By5t4qYY6ULeohizB+jiWBPQxhthlRJq
                qX7tiyUq2ClFN4kdlHXRIfRySkVKUqb3Ahh3enPTt1fWwE0q7mjs4lcuo8aZRD1r
                2eblrcfl0u1iLVae2mQYBd1RGuHB6E3QhHcrDgeRDfIVbOx7ZU/sGQ3jQU2iew5q
                13IhfI6DvOc8UbLpZhV2v7fYGUbaypEP/AtvYYIRHm/38uzE8Z/i6w6ffJ46xI5N
                8wEVOuUTXzgd16ON5LQImZ3CStcoQDEtMhJ6IWE1WtYeuTlSfqGnMF6YLJlx5Uyu
                bSMKcCdt1anx9AsMLYYfHPis7eKhKY05krikMKYb0uKyGAfLWCH073nOJTzpKj/u
                lDOSt0wqJs34GwlKSMWRkr+OvJim3yWHOrlATREt80EBGOnCmHgflc0PSkBmfvDs
                HrALD/4Ozt/DedfqLXJdNQDNP5Ji9FmBQaWun8G9iSvKLoYwskLX9RQvE0XXYWFT
                OumZCFodtUBG18nJ9aBURIgWTUeta4NZayLwgLJqInsmE8UFMdOyU8FjLW92y/E8
                3MD07LVbhh/FY4dq1ZRMpIs54/qsdHyNvJEiXwqAucK155OevaJ78nJtIyZ4qNoW
                VKBLO031iF4kP88eGOKs6lXXfTvMoI8KbqeoHWv2rlvONJ35h4FbzFbGdOnMr9jn
                Di+ybL2cxsEhL05HFDtEruMASrAO1Df//PC2CrjL0G4A2D7x7oYNfazNQ7BUZz7e
                5bvJMwI/8AcUEC3W3Wta8a0gYkls2SVlc3eUp1yiGvAV2qhKtZViQNQfihjgpQU0
                YoeeLaxcIVLKRcdWTWx1bRH5mg/IpxK9KwxJru6Gw3ltCLMXF59AQYmZDVwXjeGL
                +ziM2tNATWky
                =7dTc
                -----END PGP PUBLIC KEY BLOCK-----
            
```

muelli, 2019-08-03:
FWIW: It works much better for me when I have 
```
apt:
    sources:
        rapbian-real.list:
            source: "deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi"

```
in my user-data.

StefanScherer, 2019-08-04:
Thanks for reporting. Looks like we have to remove the debian package mirrors from the cloud.cfg

muelli, 2019-08-06:
or install the Debian keys.

PR #312 (2019-09-04): Removes Debian repos from cloud.cfg
Fixes #307 by removing the mentioned Debian Repos from cloud.cfg
closed 2019-09-04 by StefanScherer