hypriot/image-builder-rpi #308: Docker-compose not working on buster

Duvel, 2019-08-08:
I already found this with raspbian buster, but running docker-compose on hypriot 1.11 gives also the following error:

ImportError: No module named ssl_match_hostname

I found a solution here: https://github.com/docker/docker-py/issues/1502#issuecomment-506544849.

After doing it works for me:
```
sudo pip uninstall backports.ssl-match-hostname
sudo apt install python-backports.ssl-match-hostname
```

StefanScherer, 2019-09-08:
I cannot reproduce the problem. Can you share more details, eg. which docker-compose commands you called or the docker-compose.yml that you are using?

Duvel, 2019-09-08:
I think it has been resolved now in buster itself and this is no longer needed. I already noticed in a flash this week or so, that it was no longer necessary.

GabeBigBoxVR, 2019-12-03:
This is still happening to me on a fresh buster install on pi4.

Duvel, 2019-12-18:
On a fresh install this isn't an issue, but when you install pip this package apparently get's installed and then breaks docker-compose.

Maybe an idea to switch to python 3 to solve this problem.

StefanScherer, 2020-01-05:
Thanks @Duvel for the update. Yes, it's time to switch to python3.

StefanScherer, 2020-01-05:
Should be fixed with #335

closed 2020-01-05 by StefanScherer

Duvel, 2020-01-05:
Cool, are you going to do another release? I was about to flash another pi this week and could test the new release and flash tool.

StefanScherer, 2020-01-06:
There is a pre-release 1.12.0-rc2 right now where we could need some feedback.

Duvel, 2020-01-06:
I ran some test-runs and was able to break docker-compose again. But I found out that this was due to uninstalling cloud-init, which removes a few needed packages.

Can you add the packages python3-yaml and python3-jsonschema to the image? It appears that is what is needed for docker-compose.

Is there anything else that needs to be tested in this release?