hypriot/image-builder-rpi #309: hostname can not be changed using normal CLI methods

rjelbert, 2019-08-21:
I have installed hypriot 11.1.1 on a Raspberry Pi4. I used BalenaEtcher to burn the image.
I then tried to change the hostname from black-pearl to cluster-node025 - I have not been able to do this...
I tried "sudo raspi-config" change hostname option - after reboot, black-pearl was reinstated.
I tried "sudo hostname -b cluster-node025" - after reboot, black-pearl was reinstated.
I tried changing the host name in /boot/user-data. This has no impact on the hostname. The name is updated and remains but the hostname is always put back to black-pearl.
Please advise on how to change the host name via CLI.
Thanks

eloots, 2019-08-22:
@rjelbert Have you considered using [hypriot/flash](https://github.com/hypriot/flash). It has the `-n <hostname>` option the do exactly what you want to achieve.

rjelbert, 2019-08-22:
yes I have used it before but we sometimes want to change a node name or function post initial setup. The normal CLI methods worked before so this behaviour (not allowing hostname changes) is new. I am posting this in case it is a bug or perhaps a method or file that I don't know about. I have 16 RPi4s to set up and add to our cluster so checking best approach. thanks.

jeefberkey, 2019-08-22:
@rjelbert change the cloud-init script with the new hostname

MatthiasLohr, 2019-08-27:
> @rjelbert Have you considered using [hypriot/flash](https://github.com/hypriot/flash). It has the `-n <hostname>` option the do exactly what you want to achieve.

Sadly, this options doesn't seem to work for me (using newest flash script and Hypriot 1.11.1 image). The hostname given by the command line is ignored.

> @rjelbert change the cloud-init script with the new hostname

It's easier to automate with CLI parameters. Anyways: If there is such an option, I think it should work. If it's not intended to work, it should be removed.

StefanScherer, 2019-09-08:
Looks like the new cloud-init restores the hostname on every boot, so raspi-config is useless for that task. I recommend reflashing with the correct hostname configuration in the user-data file.

rjelbert, 2019-09-08:
I have solved this but it is clunky. I can confirm that the SD flash tool does set the hostname correctly. I set up 16 Raspberry Pi 4s correctly with the 1.1.11 image using the tool. However,
I did need to change the hostname on a few of them post initial set up but to do this I had to change three files to get the hostname to stick. Change:
/etc/hosts
/etc/hostname
/etc/hostname/etc/cloud/cloud.cfg ... where you set 'preserve_hostname: true'
...then reboot.

MatthiasLohr, 2019-09-10:
Sounds more like a workaround than a solution.

Maybe we should talk about variable precedence. So, what comes first, if defined. In my opinion it should be

- CLI
- cloud-init
- Raspian Defaults

That means, if a hostname (or any setting) is set via CLI, it should supersede any other definition.

Otherweise: If it's designed to be useless, please remove it from the CLI tool.

agilob, 2019-09-11:
> /etc/cloud/cloud.cfg ... where you set 'preserve_hostname: true'

this did the job, thanks. `sudo raspi-config` and changing network settings has no effect.

fdrevet, 2019-09-15:
> I have solved this but it is clunky. I can confirm that the SD flash tool does set the hostname correctly. I set up 16 Raspberry Pi 4s correctly with the 1.1.11 image using the tool. However,
> I did [...]

Thank you !! I was also struggling since yesterday with hostname change with HypriotOS 1.11...

fdrevet, 2019-09-15:
I run under Windows and flash tool is not compatible with Windows subsystem for Linux, any idea except installing a Linux VM ?

Thanks.

StefanScherer, 2019-09-16:
Just tested today again, `flash -u myhostname` works with HypriotOS 1.11.1 and newer. 
Or edit the hostname in the /boot/user-data file before first boot if you cannot use flash.