hypriot/image-builder-rpi #311: System Verification error using 1.11.1 version

thibaultserti, 2019-08-30:
Ther is many warnings that aren't really that important but infortunately I did'nt succeed to solve this issue.
Anyone has an idea ?
It occcurs when I try the init command 
`kubeadm init --pod-network-cidr 10.244.0.0/16` 

```
[init] Using Kubernetes version: v1.15.3
[preflight] Running pre-flight checks
[WARNING IsDockerSystemdCheck]: detected "cgroupfs" as the Docker cgroup driver. The recommended driver is "systemd". Please follow the guide at https://kubernetes.io/docs/setup/cri/
[preflight] The system verification failed. Printing the output from the verification:
KERNEL_VERSION: 4.19.58-v7+
DOCKER_VERSION: 19.03.1
OS: Linux
CGROUPS_CPU: enabled
CGROUPS_CPUACCT: enabled
CGROUPS_CPUSET: enabled
CGROUPS_DEVICES: enabled
CGROUPS_FREEZER: enabled
CGROUPS_MEMORY: enabled
        [WARNING SystemVerification]: this Docker version is not on the list of validated versions: 19.03.1. Latest validated version: 18.09
error execution phase preflight: [preflight] Some fatal errors occurred:
        [ERROR SystemVerification]: failed to parse kernel config: unable to load kernel module: "configs", output: "modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/4.19.58-v7+/modules.dep.bin'\nmodprobe: FATAL: Module configs not found in directory /lib/modules/4.19.58-v7+\n", err: exit status 1
[preflight] If you know what you are doing, you can make a check non-fatal with `--ignore-preflight-errors=...`
```

StefanScherer, 2019-09-16:
The following commands worked for me on HypriotOS 1.11.2

```
sudo modprobe config
sudo zcat /proc/config.gz
```

The further checks of kubeadm are out of the scope of this project. Don't know when 19.03 is in the list of validated versions.

StefanScherer, 2019-09-16:
Closing for now as it's out of scope of this project.

closed 2019-09-16 by StefanScherer