hypriot/image-builder-rpi #313: Builder fails with "E: Version '1.20190718-1' for 'raspberrypi-kernel' was not found"

troyfontaine, 2019-09-02:
When attempting to build the image (and when the CircleCI build kicks off from this repo), the error "E: Version '1.20190718-1' for 'raspberrypi-kernel' was not found" causes the build to fail.

Is there updated documentation available somewhere to help with the new process using the Raspbian original kernel?

Thanks!

StefanScherer, 2019-09-04:
That's normally a sign that the Raspberry Pi Foundation has released another kernel and we have to update that version string. See https://github.com/raspberrypi/linux/releases for current release tag names.

StefanScherer, 2019-09-04:
Fixed with #314

closed 2019-09-04 by StefanScherer

samouds, 2019-10-10:
Hello,

I am having the same issue with 1.20190819-1

StefanScherer, 2019-10-10:
@samouds looks like they just released  1.20190925-2
I tried that in my PR today morning, and had to roll back to the previous.
The version pinning is good, but not for this package, they only provide one version of the kernel 🤔

samouds, 2019-10-10:
I just tried the 1.20190925-2 and it worked

jancelin, 2019-10-24:
[1.20190925+1-1](https://github.com/raspberrypi/linux/issues/3279#issuecomment-541135632)