hypriot/image-builder-rpi #315: HypriotOS doesn't boot on raspberry pi 4B

MaartenKleijwegt, 2019-09-07:
I've just downloaded Hypriot OS 1.11.1 from https://blog.hypriot.com/downloads/ and put it on a 64gb sd card and every time I try to boot it, it hangs on the GPU test screen (the rainbow colored screen)

I've tried redownloading and checking the SD card multiple times. I've used both etcher and win32 disk imager. I've also tried flash but do not currently have access to a linux or mac system and flash doesn't work on WSL

I've also tested this SD card and pi with raspbian lite buster and everything seems to work fine. It's also not just a visual issue as the pi doesn't come online anywhere on the network.

As I'm not getting any other information from the device, I've not got much more to go on. I also can't seem to find any related issues other than that the kernel.img is probably corrupt for any generic pi OS.

Is anyone aware of any issues that might cause this?

StefanScherer, 2019-09-08:
@MaartenKleijwegt Please check if you have extracted the ZIP file

MaartenKleijwegt, 2019-09-08:
I've tried both. Etcher should be able to handle zipped files as it work's on the raspbian zipped image 

Will try setting the user-data file. It's the first time trying out hypriotos and was unaware this was required. I was assuming it works as default

StefanScherer, 2019-09-08:
It works with the defaults as well, but if you want to have different settings the best is to do it before booting the RPi.

MaartenKleijwegt, 2019-09-08:
I've changed the user-data file but still nothing works. Is there supposed to be visual output on one of the HDMI outputs or is the rainbow screen supposed to stay? My unifi network is also not reporting another device online and a network search is not reporting anything.

Also checked again with raspbian lite, but everything seems to work perfectly fine with that. Is there anything more I can try? Maybe replace the kernel.img with the working one from raspbian?

I'm getting pretty desperate right now.

*edit just popped the not working SD card into my raspberry pi 2 and it magically worked without any changes. So it must be an issue with the RPI 4. However, it works fine with regular raspbian Buster. Are there any known Issues with the 4B?

StefanScherer, 2019-09-08:
The rainbow screen must disappear as with Raspbian. 
I‘ve tested with 4B today and it works fine for me. The kernel is the same as in Raspbian.

MaartenKleijwegt, 2019-09-09:
I'm now running HypriotOS on my pi 2 and raspbian lite buster on my pi 4 (I was always planning to get HypriotOS running on both of them and maybe try my hands on Docker swarm)

However, I would like to try to get Hypriot runnning on the 4B is as well. Is there any other version than 1.11.1 working on the 4B and maybe upgrade from there? Or is there a possibility I pull in all the differences on my raspbian install? (I know this is not the general idea, but I would like to try Hypriot out and see what offers)

Legion2, 2019-09-15:
If you have one raspberry pi running you can use flash on in to create the SD card for the second pi.

StefanScherer, 2019-09-16:
Tested 1.11.1 and 1.11.2 today and it worked for me.

Duvel, 2020-01-06:
Hi, have you tried with another MicroSD card? And with another power adapter. Raspberry Pi's can be sensitive for this. I have 3 4B's and they are running all fine with Hypriot.

MaartenKleijwegt, 2020-01-09:
Yes i've  tried with another sd card. Weirdest thing was the exact same non-functional install would run perfectly fine on the raspberry pi 2. as in:

Pop newly flashed sd card into pi 4b 4gb                                                               ---> hangs on rainbow
Pop sd card out of 4b and without changing anything pop it straight  into pi 2b      ---> works perfectly fine

Tried multiple flashes on multiple sd cards on multiple power adapters, nothing worked only thing i can imagine is either it was related to the 4gb edition (Unless you guys also run that)

Or must have been something to do with mine being one of the first produced or something else that makes it different? 

Either way, I needed it for home assistant so I stopped the hassle and put raspbian lite on it and manually installed everything from there. 

Since i'm apparantly the only one suffering from this I think this can be closed and be filed as a unique unreproducable fault

Duvel, 2020-01-10:
@MaartenKleijwegt It's running fine on 2 4 GB's ones I have, so tough to say. The only thing that maybe could be different is the firmware. You can flash the EEPROM, which contains the bootloader, on the Pi 4 with a newer version.

[Raspberry Pi 4 boot EEPROM](https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md)

StefanScherer, 2020-01-10:
Thanks for sharing the Pi4 firmware update @Duvel

ayeks, 2020-01-12:
I experience the same problem on my two RaspberryPi 4b with 2GB.
Flashing the same cloud-init file onto a SD card that is used in a RaspberryPi 3 works fine. Everything in the cloud-init file got set. On the 4b it looks like it is not getting used at all.

Just flashing hypriot OS by only setting the hostname works on the 4b: `./flash -n node02 ./hypriotos-rpi-v1.12.0.img`

I will post additional information if I can narrow down the problem.

MaartenKleijwegt, 2020-01-12:
Are you sure this is the same issue? Mine never got further than the rainbow booting screen. This seems like an issue further down the line when raspbian is already speewing out info onto the screen. Seems like an easy test and something worth trying out with a second SD card.

ayeks, 2020-01-16:
Unfortunately I wasn't able to connect the pi to a screen because I have no micro-HDMI adapter. However it never connected to the network, so I assumed that it is related to the problem. Setting the hostname in the flashscript resolved that issue.

softeg, 2020-10-01:
> Unfortunately I wasn't able to connect the pi to a screen because I have no micro-HDMI adapter. However it never connected to the network, so I assumed that it is related to the problem. Setting the hostname in the flashscript resolved that issue.

Hello,
I have the same issue with my Rp4 8 go, I can't connect to it and search in my network don't detect new one.

Please you can explain me how you fix it by adding hostname ? I tried with flash script and don't work.
Thanks

Schrolli91, 2020-10-26:
same issue with RPi4B 4Gb here - doesn't boot up ...

fisruk, 2021-01-14:
I have the same problem with HypriotOS 1.11.0 on RPi4B 8GB. 

@ayeks unfortunately setting hostname on flashing does not work me.

Screen message:
`sh
start4cd.elf: is not compatible 0x0
This board required newer software. Get the latest software from: https://www.raspberrypi.org/downloads
`

mjkl-gh, 2021-01-17:
I doubt these are the same issues. As I mentioned, mine never got further than the rainbowscreen like shown in this [article](https://lifehacker.com/what-the-raspberry-pis-rainbow-boot-screen-and-rainbow-1768470271#:~:text=If%20it's%20a%20rainbow%2C%20it,supply%20as%20soon%20as%20possible.)

If you are really sure there is an issue/bug I advise you to open a new Issue. I closed this one due to running a different OS in the end which worked fine. Currently the pi in question is not in my possesion anymore so I am unable to say if this issue persists. (probably not)

mjkl-gh, 2021-01-17:
As I just received a raspberry pi 4B 8GB this week. I have decided to revisit this issue. I've flashed the pi using version 1.12.3 (1.12.0 doesn't work due to the 8GB model bootloader being too new) and got the same issue with the rainbow image again. However, this time I had a second micro HDMI cable available. Turned out the video output is only being sent over the other HDMI port. 

When encountering this issue, I was probably facing a misconfiguration of the network settings and never bothered to check them as hooking a screen up produced the rainbow image. I think I remember checking both ports, but it's a year ago so it might not be.

Also tested a regular raspberry pi OS image, for those both video outputs work. There is probably something with the cmdline.txt that configures a single hdmi output. I've pulled both from hypriot and Pi OS:

```
Hypriot
dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=c73ba28a-02 rootfstype=ext4 cgroup_enable=cpuset cgroup_enable=memory swapaccount=1 elevator=deadline fsck.repair=yes rootwait quiet init=/usr/lib/raspi-config/init_resize.sh
```
```
Pi OS
dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=7f86517d-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
```

@StefanScherer Can you maybe confirm this is working as intended? Or is it a side-effect of something else?

TLDR: hypriot outputs HDMI to only 1 output on a raspberry pi 4b and outputs the rainbow image to the other. The pi not showing up on the network is/was probably a networking issue.

davidchase, 2022-01-16:
i had the same as issue as you @mjkl-gh where i only saw the rainbow screen for each time i tried to flash any version of hypriot. and confirmed that my other HDMI port shows what is expected and the original port still shows rainbow image.