hypriot/image-builder-rpi #319: Cloud-init ssh-import-id fails

troyfontaine, 2019-09-17:
Using the PR on flash that updates the date and time to when the SD card is created, I attempt to use ssh-import-id however it fails with the following error that I was able to pull from the cloud-init logs:
```
Cloud-init v. 18.3 running 'modules:config' at Tue, 17 Sep 2019 03:47:37 +0000. Up 76.03 seconds.
2019-09-17 03:47:38,643 - util.py[WARNING]: Failed to run command to import pirate ssh ids
2019-09-17 03:47:38,657 - util.py[WARNING]: ssh-import-id failed for: pirate ['gh:myusername']
2019-09-17 03:47:38,659 - util.py[WARNING]: Running module ssh-import-id (<module 'cloudinit.config.cc_ssh_import_id' from '/usr/lib/python3/dist-packages/cloudinit/config/cc_ssh_import_id.py'>) failed
```

I can manually run `ssh-import-id` from the shell without issue after it completes cloud-init.

Running `sudo cloud-init collect-logs` gives me some more to go by, I think:

```
2019-09-17 03:47:38,643 - util.py[DEBUG]: Failed to run command to import flynn ssh ids
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/cloudinit/config/cc_ssh_import_id.py", line 104, in import_ssh_ids
    util.subp(cmd, capture=False)
  File "/usr/lib/python3/dist-packages/cloudinit/util.py", line 2027, in subp
    cmd=args)
cloudinit.util.ProcessExecutionError: Unexpected error while running command.
```

and

```
Traceback (most recent call last): 
File "/usr/lib/python3/dist-packages/cloudinit/config/cc_ssh_import_id.py", line 79, in handle                            import_ssh_ids(import_ids, user, log)
File "/usr/lib/python3/dist-packages/cloudinit/config/cc_ssh_import_id.py", line 107, in import_ssh_ids
raise exc
File "/usr/lib/python3/dist-packages/cloudinit/config/cc_ssh_import_id.py", line 104, in import_ssh_ids
util.subp(cmd, capture=False)
File "/usr/lib/python3/dist-packages/cloudinit/util.py", line 2027, in subp                                               cmd=args)
cloudinit.util.ProcessExecutionError: Unexpected error while running command.
```

So all that I can speculate is that even with the time stamp being much closer to accurate with the updated flash utility, the time drift is still enough to prevent ssh-import-id from working without a proper ntp sync.

Legion2, 2020-01-12:
The original PR was closed in favor of a better solution (hypriot/flash#162) and was released in v1.12.0 and version 2.5.0 of the flash tool. You may want to try if it also fixes your issue.

troyfontaine, 2020-01-16:
Thanks @Legion2 I'll close this issue (It looks like a solid fix-I just won't have a chance to test it for a bit).

closed 2020-01-16 by troyfontaine

verglor, 2020-05-27:
I have the same problem after flashing [hypriotos-rpi-v1.12.1.img](https://github.com/hypriot/image-builder-rpi/releases/download/v1.12.1/hypriotos-rpi-v1.12.1.img.zip). 

I don't use [hypriot/flash](https://github.com/hypriot/flash) - I directly flash image in windows with [etcher](https://www.balena.io/etcher). However it is not working even if I manually set `fake-hwclock.data` to current UTC time in boot partition.

Cannot `ssh-import-id` be invoked only after correct time is set by NTP without needing to use specific flash utility or manually updating `fake-hwclock.data` ?