hypriot/image-builder-rpi #320: Built-in Bluetooth does not start on a Raspberry Pi 4 4GB

androidand, 2019-09-19:
I've tried flashing hypriotos-rpi-v1.11.2 and 1.11.0 with the same result on a brand new Raspberry Pi 4, 4GB. The build in bluetooth does not start.

`sudo systemctl status hciuart.service -l
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/etc/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-09-16 09:26:27 CEST; 3 days ago
  Process: 382 ExecStart=/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow - (code=exited, status=1/FAILURE)

Sep 16 09:25:57 gunmetal systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Sep 16 09:26:27 gunmetal hciattach[382]: Initialization timed out.
Sep 16 09:26:27 gunmetal hciattach[382]: bcm43xx_init
Sep 16 09:26:27 gunmetal systemd[1]: hciuart.service: Control process exited, code=exited, status=1/FAILURE
Sep 16 09:26:27 gunmetal systemd[1]: hciuart.service: Failed with result 'exit-code'.
Sep 16 09:26:27 gunmetal systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.`

Naturally no devices showed up when running hcitool dev.

I tried a lot of things on 1.11.0 like updating everything, basically trying all tips I could find. I then saw there was a newer release out, 1.11.2 (the first one downloaded from Hypriot webpage, the second from github). Anyway, after installing and running 1.11.2, I found the same issue, bluetooth service was not running. However, this time, fixing it was as simple as starting it:

`sudo systemctl status hciuart.service -l
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/etc/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-09-19 19:29:13 CEST; 3s ago
  Process: 1221 ExecStart=/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow - (code=exited, status=0/SUCCESS)
 Main PID: 1230 (hciattach)
    Tasks: 1 (limit: 4915)
   Memory: 792.0K
   CGroup: /system.slice/hciuart.service
           └─1230 /usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -

Sep 19 19:29:06 gunmetal systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Sep 19 19:29:13 gunmetal hciattach[1221]: bcm43xx_init
Sep 19 19:29:13 gunmetal hciattach[1221]: Flash firmware /lib/firmware/brcm/BCM4345C0.hcd
Sep 19 19:29:13 gunmetal hciattach[1221]: Set Controller UART speed to 921600 bit/s
Sep 19 19:29:13 gunmetal hciattach[1221]: Device setup complete
Sep 19 19:29:13 gunmetal systemd[1]: Started Configure Bluetooth Modems connected by UART.`

`hcitool dev
Devices:
	hci0	DC:A6:32:21:D2:88`

Question, shouldn't the service start automatically? 
Anyway, perhaps this helps someone!

Naboleo, 2019-09-22:
Hello,

Got the same problem here. 
Actually service is starting automaticaly : status is OK with systemctl right after boot sequence. But few secondes later, it stops.... This doesn't occur with buster-lite 2019-07-10

Naboleo, 2019-11-25:
Applyed image 1.11.4 and this
https://github.com/hypriot/image-builder-rpi/commit/36d125bf07b95b86fe809b91a00bd26a7a8f7097

ATM everything is fine with BT

@StefanScherer shouldn't it already part of the images ?