hypriot/image-builder-rpi #324: macvlan; operation not supported?

talondnb, 2019-10-15:
Has anyone managed to successfully launch a container using macvlan? I've spent hours trying this with different container releases and I keep getting:

docker: Error response from daemon: failed to create the macvlan port: operation not supported.

Running latest hypriot on a Pi 3B. Any advice welcome!

talondnb, 2019-10-15:
I've managed to figure out that I need to enable experimental features:

https://github.com/hypriot/docker/blob/master/experimental/vlan-networks.md

rezadarzi, 2020-04-27:
> 
> 
> I've managed to figure out that I need to enable experimental features:
> 
> https://github.com/hypriot/docker/blob/master/experimental/vlan-networks.md

can you solve this problem?
I have the same problem with ipvlan and also experimental feature is on.

talondnb, 2020-04-27:
Yes, I'm running macvlan on my unifi-controller docker container.

Here's my docker-compose.

```
version: "3.6"

services:
  unifi-controller:
    image: linuxserver/unifi-controller:latest
    container_name: unifi-controller
    networks:
      default:
        ipv4_address: 10.10.1.10
    environment:
      - PUID=1000
      - PGID=1000
      - MEM_LIMIT=1024M #optional
    volumes:
      - ./config:/config
    ports:
      - 3478:3478/udp
      - 10001:10001/udp
      - 8080:8080
      - 8081:8081
      - 8443:8443
      - 8843:8843
      - 8880:8880
      - 6789:6789
    restart: unless-stopped

networks:
  # Define the default network
  default:
    external:
      name: macvlan1

```
And here's my portainer network settings for macvlan (sorry not sure where I would see this in CLI as I set this up through portainer.

[image]

rezadarzi, 2020-04-27:
> Yes, I'm running macvlan on my unifi-controller docker container.
>
> Here's my docker-compose.
>
> ```
> version: "3.6"
>
> services:
>   unifi-controller:
>     image: linuxserver/unifi-controller:latest [...]

thank you for your attention.
you don't have the swarm cluster, isn't it?
how your problem solves?

talondnb, 2020-04-27:
No, not using a swarm.

I simply wanted my unifi container to be on the same physical segment as my other devices and not a virtual segment. This worked out fine for me.