hypriot/image-builder-rpi #325: Edit Cloud-Init after first boot

bd8392, 2019-11-01:
Hi,

I don't know if this is even possible but I tried to edit the user-data under /boot after the very first boot of hypriot, but it didn't take effect...is there another way? Because I want to test multiple cloud-init possibilities without always set up a new machine...

bd8392, 2019-11-02:
Found it out! With the command: "sudo cloud-init clean" you can tell cloud init to run at next reboot =)

closed 2019-11-02 by bd8392

StefanScherer, 2019-11-02:
Thanks @bd8392 for sharing the solution!

bd8392, 2019-11-03:
@StefanScherer no problem :-) but be careful when using it. When I got it right, it doesn't revert the configs that the previous Cloud-init made, but applies the new ones over it...so you maybe get mixed up with settings when rerunning cloud-init...