hypriot/image-builder-rpi #329: how to compile kernel modules for a wireless adapter with hypriotOS

altineller, 2019-11-27:
Hello,

I am trying to make a USB wireless adapter (EW-7822) to work with hypriotOS.

I have done this before both on ubuntu and raspberry pi, so it can be done.

How would I approach to build and install a dkms kernel module? The outline is given in https://edimax.freshdesk.com/support/solutions/articles/14000041287-how-to-install-ew-7811-ac600-ew-7822uac-in-linux-with-kernel-higher-than-v4-1 

But since hypriotOS does not have build environment, I could not do it.

There is also a ready kernel module called rtl8812au, which is the same chipset, but however it is locked by usb id. [it is the same chipset, but the module does have certain usb ids hard coded and it will not work with the edimax]

I am using hypriotOS for ros, and I need better communications, so I need this usb3.0 wireless adapter working.

Best Regards,
C.A.

StefanScherer, 2019-11-27:
Which version of HypriotOS do you use?
The last couple of releases we use the upstream raspbian kernel, so maybe just a package is missing. 
Could you check installed packages on raspbian and hypriot?

altineller, 2019-11-29:
Hello,

I am using the latest, 1.11.0.
 using: # apt search rtl8822 we get:

firmware-realtek/testing 1:20190114-1+rpt4 all [upgradable from: 1:20190114-1+rpt2]
  Binary firmware for Realtek wired/wifi/BT adapters

which is the required module actually, and it is contained within hypriot os, but however,
The EW-7822 device has the following usbid:

>  lsusb | grep Edimax
Bus 003 Device 003: ID 7392:c822 Edimax Technology Co., Ltd

The firmware-realtek package has the module, but since it is customized for realtek devices, it will not recognize the usbid 7392:c822 that belongs to edimax ew-7822

best regards,
Can