hypriot/image-builder-rpi #336: wpa-supplicant regularly fails to initialize

altineller, 2020-01-06:
Hello,

I have a problem of wifi not associating with the AP, in some of the boots. It is configured and it works, but in some of the time the wifi will not associate. I have to connect the ethernet, then manually delete /var/run/wpa-supplicant directory, and then reboot, and wifi will work.

Is there a solution to this problem?

StefanScherer, 2020-01-07:
Hello @altineller, 
which version do you use?
I saw this problem from time to time. In my case I just unplugged power cable, plugged it in again (so the RPi booted a second time) and then WiFi just works without logging into it.
I only have a screenshot from my error message. I'll add it here to track some progress to investigate the root cause.

[image]

```
wpa_supplicant: /sbin/wpa_supplicant daemon failed to start
run-parts: /etc/network/if-pre-up.d/wpasupplicant exited with return code 1
ifup: failed to bring up wlan0
```

altineller, 2020-01-08:
Hello,

I am using the hypriotos-rpi-v1.11.0.img

Yes, it sometimes goes away if you power cycle. But I am using this for a robot, where the hardware is not always accessible, and this is a showstopper bug for me. Each time I turn the robot on, there is 50/50 chance it will not connect to network.

Best regards,
C.

ProfFan, 2020-01-18:
@altineller Are you also trapped by the Duckiebot 🤣 Same here...

altineller, 2020-01-20:
> @altineller Are you also trapped by the Duckiebot rofl Same here...

@ProfFan 

not quite, but this problem probably repeats in duckiebot, although they are using v1.9.0 of hypriot.

I think the problem is not related with hypriot, but rather raspbian, or even the raspberry pi hardware.

huntipl, 2020-05-04:
Hi, I',m seeing the same behaviour as described above. 
After looking at syslog I can see that wpa_supplicant fails due to:
`/etc/network/if-pre-up.d/wpasupplicant: 120: /etc/network/if-pre-up.d/wpasupplicant: cannot create /dev/stderr: No such device or address`. Any ideas how to address this?