hypriot/image-builder-rpi #341: Raspberry Pi 4 + POE Hat: stuck after the second rebooot

fwiedmann, 2020-01-20:
**This issue only belongs to hypriot os. Could not reproduce it with raspbian buster lite.**

**Describe the bug**
After rebooting the second time a raspberry pi 4 4GB with PEO hat it will not start successfully. It's not possible to login via SSH (Destination Host Unreachable). The system is configured with a static ipv4 address.
Used POE Switch: TP-Link TL-SG10I05P 5-Port Gigabit PoE Switch. 

I can reproduce this behaviour on 4 raspberry pi with the same setup. After re-plug the ethernet cable for a manual reboot the pi starts successfully again.

**To reproduce**
Reboot the system several times in a row. Try to login with SSH or to ping the system. 

**Expected behaviour**
To be able to reach the system via ping or ssh.

**Actual behaviour**
After a second reboot its not possible to reach the system via ssh or ping.
The LEDs on the pi ethernet interface does not blink like before when it has an link to the network.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
```
Pi4B 4GB
```

* Which OS and version (`cat /etc/rpi-issue`)?
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
HYPRIOT_DEVICE="Raspberry Pi"
HYPRIOT_IMAGE_VERSION="v1.12.0"
```

* Which firmware version (`vcgencmd version`)?
```
Copyright (c) 2012 Broadcom
version d844d7ce19c1696e0f6110aab106e250c3099147 (clean) (release) (start_cd)
```

* Which kernel version (`uname -a`)?
```
Linux k3s-cluster-worker-0 4.19.95-v7l+ #1292 SMP Thu Jan 16 15:50:13 GMT 2020 armv7l GNU/Linux
```
**Logs**
If applicable, add the relevant output from `dmesg` or similar.

```
 0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.95-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1292 SMP Thu Jan 16 15:50:13 GMT 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 64 MiB at 0x000000003a400000
[    0.000000] On node 0 totalpages: 1025536
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 828928 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s36928 r8192 d24512 u69632
[... 257 lines cut ...]
[    2.990496] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[    2.993521] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    2.995486] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[    3.000014] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[    3.037031] bcm2835_audio soc:audio: card created with 8 channels
[    3.096283] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    3.243761] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    3.316607] brcmfmac: F1 signature read @0x18000000=0x15264345
[    3.327389] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.328123] usbcore: registered new interface driver brcmfmac
[    3.556833] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.590042] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    4.547890] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   10.011136] uart-pl011 fe201000.serial: no DMA platform data
[   10.182520] 8021q: 802.1Q VLAN Support v1.8
[   10.374739] bcmgenet: Skipping UMAC reset
[   10.472169] bcmgenet fd580000.genet: configuring instance for external RGMII (no delay)
[   10.500657] brcmfmac: power management disabled
[   11.512010] bcmgenet fd580000.genet eth0: Link is Down
[   15.672141] bcmgenet fd580000.genet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   58.071935] random: crng init done
[   58.071946] random: 7 urandom warning(s) missed due to ratelimiting
[   58.587872] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
[   58.593481] Bridge firewalling registered
```

Legion2, 2020-02-22:
I have the same issue with Raspberry Pi 3 B and B+.
After the first reboot the raspberry pi did not start correctly and I had to remove the usb power. After powering it on again, it started correctly but had a different SSH Host key then before the restart.

Everything else works as expected now.

I used hypriot version v1.12.0 and flash version 2.5.0.
```
$ uname -a
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
```

r1cebank, 2020-03-11:
I have the same issue, multiple restarts works, but host key changed. Using Raspberry pi 4b 4GB