hypriot/image-builder-rpi #342: Docker IPv6 DNS failure on boot

Vynce, 2020-01-27:
Once the Pi boots up, running any docker commands that interact with the network fails like this:
```
$ docker search foo
Error response from daemon: Get https://index.docker.io/v1/search?q=foo&n=25: dial tcp: lookup index.docker.io on [::1]:53: read udp [::1]:38014->[::1]:53: read: connection refused
```

If I try again, it'll continue failing for a few seconds and will then start working:
```
$ docker search foo
NAME                                         DESCRIPTION                                     STARS               OFFICIAL            AUTOMATED
foomo/pagespeed_exporter                     Pagespeed Exporter Build                        3                                       [OK]
```

It'll continue working until the Pi is rebooted. After rebooting it'll fail the same way.

If I disable IPv6 support by adding `ipv6.disable=1` to `/boot/cmdline.txt` it just fails with a different error:
```
$ docker search foo
Error response from daemon: Get https://index.docker.io/v1/search?q=foo&n=25: dial tcp: lookup index.docker.io on [::1]:53: dial udp [::1]:53: socket: address family not supported by protocol
```

This issue only seems to affect docker, so it might be related to go's hostname resolution implementation? I've looked through the logs and tried all kinds of tweaks to track down the root cause. It seems like something is causing docker to initially try resolving hostnames via IPv6. That fails/times out after a while and then IPv6 isn't used anymore.

Restarting `dockerd` doesn't reset the behavior. Restarting `systemd-resolved` empties out `/etc/resolv.conf` which makes all hostname resolution fail. Restarting `dhcpd` repopulates `/etc/resolv.conf` correctly. Docker search still works after that, so I don't what else could be storing this dns/IPv6 state. Maybe the kernel?

Environment
- Hypriot 1.12.0
- user-data example from https://github.com/hypriot/flash#cloud-init
- Wired IPv4 network

Vynce, 2020-02-04:
I installed docker-ce on Raspbian buster and don't have the same DNS issue there. systemd-resolved is disabled in Raspbian and it uses openresolv instead.

```
Raspbian
/etc/resolve.conf generated by resolveconf
systemd-resolved.service; disabled

Hypriot
/etc/resolve.conf generated by dhcpd
systemd-resolved.service; enabled
```

I disabled systemd-resolved and installed openresolv on Hypriot and, unsurprisingly, still have the same DNS issue.

I don't see any significant differences in `ifconfig` and `sysctl -a` between both OSes.

I grabbed an [strace log of dockerd](https://github.com/hypriot/image-builder-rpi/files/4150787/dockerd_strace.log) trying to connect to ::1 port 53, but still no clues about why it thinks something is listening on ::1 port 53 and why connecting to that would be a good idea.

Legion2, 2020-03-07:
I have the same. After boot I connect via ssh and try to pull docker images but I get some type of dns lookup error.

Joshua-Anderson, 2020-03-27:
Glad I'm not insane! I'm seeing this exact same issue bizzare as well.
I'm on an IPv6 network, but my resolv.conf definitely doesn't include localhost.
```
# Generated by dhcpcd from eth0.dhcp, eth0.ra
# /etc/resolv.conf.head can replace this line
nameserver 10.42.43.1
nameserver 2606:4700:4700::1111
```

Happens on all 3 raspberry pis (rpi 3 and 3+) in my cluster. First docker command fails with `::1` dns lookup, wait 30 seconds afterwards and everything starts working fine.

```
$ docker pull hello-world
Using default tag: latest
Error response from daemon: Get https://registry-1.docker.io/v2/: dial tcp: lookup registry-1.docker.io on [::1]:53: read udp [::1]:47412->[::1]:53: read: connection refused
```

Tried overriding docker daemon dns server with the following config and had no luck.

```
{
  "log-driver": "journald",
  "dns": [ "1.1.1.1" ]
}
```

Seems vaguely related to minio/mc#2564, another arm go program getting the exact same error.
Seems likely the issue lies with the golang resolver, not docker, but that doesn't explain why raspbian docker appears fine.

I don't know enough about golang dns resolution to really know where to go from there, but am happy to help debug or keep researching if somebody can point me in right direction