hypriot/image-builder-rpi #346: Increase /boot Partition Size

mxklb, 2020-02-14:
Actually I can't install the 2mic [respeaker](https://github.com/respeaker/seeed-voicecard) driver on a vanilla flashed HypriotOS v1.12.0 on a PI4. What happens is the /boot partition of the OS is too small (~12MB free space) to downgrade the kernel during seeed-voicecard install.sh script. So far so good, finally I end up with a bricked HypriotOS with read-only file system, which is seeed-voicecards fault, therefore see https://github.com/respeaker/seeed-voicecard/issues/209

HypriotOS shall have enought space on /boot to downgrade kernels, so please 

- [ ] increase the initial size of the /boot partition 

to be at least 100MB.

https://serverfault.com/questions/334663/what-is-the-recommended-size-for-a-linux-boot-partition

After flashing HypriotOS v1.12.0 from github releases to a 64GB card:
```
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        56G  1.1G   53G   2% /
devtmpfs        1.9G     0  1.9G   0% /dev
tmpfs           2.0G     0  2.0G   0% /dev/shm
tmpfs           2.0G   17M  2.0G   1% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           2.0G     0  2.0G   0% /sys/fs/cgroup
/dev/mmcblk0p1   64M   52M   12M  82% /boot
tmpfs           396M     0  396M   0% /run/user/1000
```

mxklb, 2020-02-18:
may be linked to #347

StefanScherer, 2020-02-20:
Maybe this is not a good use-case for HypriotOS. I would suggest to use standard Raspbian, install everything you need and install Docker with the get.docker.com install script.

mxklb, 2020-02-20:
What exactly is the usecase for HypriotOS? Maybe I missed that .. 

I just want to run docker containers. Ok, for audio card access I need Hardware.. but isn't HypriotOS on a PI a good idea if I'd like to plug additional Hardware which needs custom drivers? I don't understand your point, maybe you could clarify a little.. 

Actually I already investigated and tried some other OS because of this, these seem to work better.

P.S. My "usecase" was successfully deployed to HypriotOS with using an older version mid 2019!

StefanScherer, 2020-03-11:
I don't know if replacing the kernel still has all relevant features needed to use Docker. That's why I suggested to use Raspbian for such projects. The idea of Hypriot is to run Docker and we cannot support other kernels installed if something breaks.
Well if it works, fine. :-) 

Increasing /boot should be done here: https://github.com/hypriot/image-builder-raw/blob/master/builder/rpi/build.sh#L13 and drafting a new release. We want to keep the partitions as small as possible and useful.

StefanScherer, 2020-03-13:
I checked, Raspbian has a 256 MB /boot partition. It's not the size I would prefer.
Would 100 MB be enough @mxklb ?

mxklb, 2020-03-15:
Hi Stefan, thx, I'll check the image-builder if I find time to (don't know when) .. 

I also don't like the kernel to be downgraded. So I think there is no big need to support it, that's fine. The kernel downgrade of the driver I'm using looks like this:
```
/tmp/raspberrypi-kernel_1.20190925+1 100%[====>]  67.38M  3.29MB/s    in 28s

/tmp/raspberrypi-kernel-headers_1.20 100%[=====>]  23.74M  2.62MB/s    in 9.3s
```
and it errors with: *You do not have enough space in /boot to install this package*. You can find full error log of kernel dowgrade in the linked issue https://github.com/respeaker/seeed-voicecard/issues/209

Actually I don't know how much space the/a kernel downgrade needs. If it's the size of the downloaded kernel itself, 100MB would not be enough. And if kernel headers download size is needed aswell, you should better increase to 150MB or more .. tbd

I can reproduce/test it with different boot volume sizes if you provide me an image download link.

mxklb, 2020-03-16:
I checked the Raspbian version I have installed this driver to. After driver installation, kernel downgrade and reboot, it still consumes only 52MB at /boot
```
pi@raspberrypi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        29G  9,6G   19G  35% /
devtmpfs        1,8G     0  1,8G   0% /dev
tmpfs           2,0G  6,0M  2,0G   1% /dev/shm
tmpfs           2,0G  9,1M  1,9G   1% /run
tmpfs           5,0M  4,0K  5,0M   1% /run/lock
tmpfs           2,0G     0  2,0G   0% /sys/fs/cgroup
/dev/mmcblk0p1  253M   52M  201M  21% /boot
tmpfs           391M     0  391M   0% /run/user/1000
```
The missing space for downgrade may only be used temporarily until next reboot. But I don't know ..

Maybe you could find out how much space a typical downgrade needs. So you could feature but  not support projects running docker on the edge with specific kernel versions. Trade off between projects that need minimal /boot volume size and choose default HypriotOS Image /boot volume size depending on this decision. Nevertheless, thank you for your support!

mxklb, 2020-03-16:
> may be linked to #347

Means that I am not sure if the respeaker driver installation caused #347 somehow, tbd

FoxRomeo, 2020-08-03:
rpi-update with todays kernel/repository:

# rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Backing up firmware
 *** Backing up modules 5.4.51-v7+
Partition size 64M may not be sufficient for new Pi4 files
This could result in a system that will not boot.
256M FAT partition is recommended. Ensure you have a backup if continuing.
Would you like to proceed? (y/N)