hypriot/image-builder-rpi #355: make sd-image fails: "E: Version ... for 'raspberrypi-kernel' was not found"

ctrahey, 2020-05-01:
Fresh clone of this repo (tag v1.12.0), fresh install of Vagrant (2.2.7) and VirtualBox (on my host Mac - 6.1.6).

On `make sd-image` (after `vagrant up`), I get the following error (full output attached).

I've toyed with the version a little in versions.config, but I don't know which repo is referenced, so I'm not sure how to find a valid version number. 

```
...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Processing triggers for initramfs-tools (0.133+deb10u1) ...
+ '[' -z '' ']'
+ apt-get install -y --no-install-recommends raspberrypi-kernel=1.20190925+1-1
Reading package lists...
Building dependency tree...
Reading state information...
E: Version '1.20190925+1-1' for 'raspberrypi-kernel' was not found
make: *** [sd-image] Error 100
```

[output.txt](https://github.com/hypriot/image-builder-rpi/files/4562177/output.txt)

Long term, I'm attempting to customize the kernel a bit, but for now I can't get a vanilla build happening. Thanks in advance for any insights!

ctrahey, 2020-05-02:
Okay, I poked around a bit and found that it needed the version bumped to `KERNEL_BUILD="1.20200212-1"`

Seems like this is not actually an issue - apologies for the false alarm. Might be worth making a note to future readers about how to discover the latest version of this package, or where to point sources to grab an older version.

mattvonrocketstein, 2020-06-22:
Yes, how *do* we discover the latest version of this package or find older versions?