hypriot/image-builder-rpi #359: Setting static ip address in v1.12.1

rbartoli, 2020-05-23:
I did flash v1.12.1 after editing `/boot/user-data` with the following content:

`#cloud-config
# vim: syntax=yaml
#

hostname: hypriot
manage_etc_hosts: true

users:
  - name: pirate
    gecos: "Hypriot Pirate"
    sudo: ALL=(ALL) NOPASSWD:ALL
    shell: /bin/bash
    groups: users,docker,video,input
    plain_text_passwd: hypriot
    lock_passwd: false
    ssh_pwauth: true
    chpasswd: { expire: false }

locale: "en_GB.UTF-8"
timezone: "Europe/London"
package_upgrade: false

write_files:
  - content: |
      persistent
      # Generate Stable Private IPv6 Addresses instead of hardware based ones
      slaac private
      # static IP configuration:
      interface eth0
      static ip_address=192.168.0.5/24
      # static ip6_address=fd51:42f8:caae:d92e::ff/64
      static routers=192.168.0.1
      static domain_name_servers=192.168.0.1 8.8.8.8
    path: /etc/dhcpcd.conf
runcmd:
  - 'systemctl restart avahi-daemon'`

However the static ip configuration seems to have no effect. The RPI boots properly, I can log into the system, however the ip address of the box is not 192.168.0.5.

I did re-flash the same SD card multiple times and manually checked `/etc/dhcpcd.conf` and it contains the expected content.

So I did try to use the same above configuration on **v1.12.0**, flashed it on the same SD and it worked as expected.

dennym, 2020-06-17:
I figured it out.
Basically since dhcp was deactivated by #247  I tried to reactivate it and use the example to set the static IP but this led to multiple different issues and multiple appearing IP addresses in the network.

I managed then to set the static IP address with cloud init following this guide:
https://clientarea.ramnode.com/knowledgebase/4180/Static-IP-Configuration-Cloud.html
Using the debian example works. I think this was the intended way for the PR but didn't get documented or mentioned.

StefanScherer, 2020-06-17:
That's awesome @dennym that you found a solution.

Do you want to share it with a small PR to https://github.com/hypriot/flash/blob/master/sample/static.yml to show the correct configuration? Or if it's useful for more people add it to the FAQ https://github.com/hypriot/blog/blob/master/content/faq.md 🤗

Thanks!

rbartoli, 2020-06-17:
@dennym Amazing! If you could even just share your configuration file, that will be great.

dennym, 2020-06-17:
Unfortunately there is no script. I manually added the configuration mention in the link to each node. But I am sure you can utilize the `write_files` part where you did the dhcp config to set the cloudinit configuration.

highvight, 2020-07-03:
I'm glad this has been brought up (see also #366 ). Apparently there are various ways to set up static ip adresses. Using the dhcpcd daemon is recommended since Raspbian Jessie, but is deactivated in hypriot by default (as @rbartoli pointed out). 

Would be great if we came up with some best practice here.

616b2f, 2020-07-17:
@rbartoli try this:

write to file `network-config`
```yaml
version: 1
config:
  - type: physical
    name: eth0
    subnets:
      - type: static
        address: 172.16.1.21/24
        gateway: 172.16.1.1
        dns_nameservers: 
          - 172.16.1.1
          - 8.8.8.8
        dns_search:
          - clmaster01
```
and use it like this:

```sh
$ flash -u user-data.yml -F network-config hypriotos.img.zip
```
There is some time past but I remember that this worked for me when I setup my dev rpi cluster at home.

Let me please know if it works for you.

highvight, 2020-08-31:
This works for me:
`flash -F network-config hypriotos-rpi-v1.12.3.img`

Example for network-config:
```
version: 2
ethernets:
  eth0:
    addresses:
      - 192.168.1.100/255.255.255.0
```

Notes:
- Name the network config yaml exactly `network-config`
- Refer to these docs for [version 1](https://cloudinit.readthedocs.io/en/latest/topics/network-config-format-v1.html) and [version 2](https://cloudinit.readthedocs.io/en/latest/topics/network-config-format-v1.html). **Remove the top network key from the examples!**

### What's going on?

To load user-data, meta-data and network-config, Hypriot is using NoCloud as datasource. Read more about this in the cloud-init [documentation](https://cloudinit.readthedocs.io/en/latest/topics/datasources/nocloud.html). Most importantly, there are different ways to define network configurations and using a network-config.yaml file requires you to not use the top network key.

Behind the scenes it is actually pretty simple. Look at the following files after flashing and mounting the sdcard:

- In `HypriotOS/` you can find the meta-data, user-data and network-config files. Hypriot has default files for these, but you can load different files with the flash tool and the corresponding flags. You can also change these files manually, of course, but before booting.
- In `root/var/lib/cloud/seed/nocloud-net/` you can find links to these files from the root filesystem. Cloud-init is using these files to configure all sorts of stuff on first boot, for instance `root/etc/network/interfaces.d/50-cloud-init.cfg` for network interfaces.

Unfortunately, the flashing tool has no specific flag for importing network-config file. That's why we need to use -F as a general way to import files (and that's why the naming is important). 

One last note: If you want to alter the network configurations after first boot, you will need to restart cloud-init one way or the other. [This](https://cloudinit.readthedocs.io/en/latest/topics/cli.html) could work:
```
cloud-init clean
reboot
```

ajaegle, 2020-11-12:
Am I right, that applying just the config for eth0 (as seen above) will remove the docker0 interface and the network bridge, so that accessing docker services via network isn't working any longer? Did somebody come up with a network-config that keeps that setup in place?