hypriot/image-builder-rpi #361: Invalid `authorized_keys` file when `ssh_authorized_keys` specified in `user-data`

verglor, 2020-05-27:
Steps to reproduce:
1. Flash [hypriotos-rpi-v1.12.1.img](https://github.com/hypriot/image-builder-rpi/releases/download/v1.12.1/hypriotos-rpi-v1.12.1.img.zip) to sdcard
2. append the following to `user-data` on boot partition:
```
ssh_authorized_keys:
    - ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIPDz/lROfK9tdARDHTMPH/vU6MmkpcAfMNViaigHyqPy verglor
```
3. boot with sdcard and wait for `cloud-init` to finish
4. there is no `/home/pirate/.ssh/authorized_keys`
5. there is `/root/.ssh/authorized_keys` that contains invalid content:
```
no-port-forwarding,no-agent-forwarding,no-X11-forwarding,command="echo 'Please login as the user \"NONE\" rather than the user \"root\".';echo;sleep 10" ssh-ed25519 AAAAC3NzaC1lZDI1NTE5AAAAIPDz/lROfK9tdARDHTMPH/vU6MmkpcAfMNViaigHyqPy verglor
```

StefanScherer, 2020-05-27:
Hm, the following `user-data` works for me.

```yaml
#cloud-config
# vim: syntax=yaml
#

# Set your hostname here, the manage_etc_hosts will update the hosts file entries as well
hostname: black-pearl
manage_etc_hosts: true
# don't write debian.org into apt mirrors
apt_preserve_sources_list: true

# You could modify this for your own user information
users:
  - name: stefan             # use any user name you like
    primary-group: users
    shell: /bin/bash
    sudo: ALL=(ALL) NOPASSWD:ALL
    groups: users,docker,adm,dialout,audiolugdev,netdev,video
    ssh-import-id: None
    lock_passwd: true
    ssh-authorized-keys:
      - ssh-rsa AAAA.....NN stefan
```

oneirosFade, 2020-09-19:
Similar issue (not specific to SSH so I can create a separate issue if required) where user-data appears to be completely ignored. Something as simple as changing the hostname parameter (either before first boot with a fresh user-data file, or post-first-boot by modifying the present user-data file and rebooting) doesn't take.