hypriot/image-builder-rpi #362: There is a new 8GB variant of the Raspberry Pi 4

Legion2, 2020-05-28:
The [8GB variant](https://www.raspberrypi.org/blog/8gb-raspberry-pi-4-on-sale-now-at-75/) also uses a 32-bit Kernel which can address all 8GB but allow the userspace applications to use up to 3GB per process. Are there any changes required to use the 8GB variant with hypriot or should it work as is?

StefanScherer, 2020-05-28:
Thanks @Legion2 for the heads up. I read the news today as well. Very exciting.
I assume the 8GB variant works with current HypriotOS just fine. Let us know if you grab one and give it a try.

geerlingguy, 2020-06-02:
I may give this a try, but on top of the initial request—will HypriotOS be making the shift to be arm64? Or is the intention to keep it 32-bit for now? There are a lot more Docker images available with arm64 builds than there are armv7, especially from official sources, and it would be nice to be able to run with arm64.

Note that in my preliminary testing, switching to the Pi OS 64-bit version for any Pi with < 4 GB of RAM (and for apps which need lots of threads that use as much memory as they can get), it's actually slower under 64-bit than 32-bit because less RAM is available due to larger pointers for small bits of memory...

Also I'm assuming this switch would basically break support for any Pi older than the Pi model 3 B?

StefanScherer, 2020-06-03:
The priority is the 32bit OS to support all Raspberry Pi devices, especially as long as the small Pi Zero form factor is only 32bit.

I've tested the 64bit Ubuntu distro which is excellent and well supported.

geerlingguy, 2020-06-14:
@StefanScherer - Just as a note, I didn't have any trouble with the current HypriotOS on the 8GB model of Pi 4. I've been testing it against both the 32-bit and 64-bit Pi OS versions, and while many operations are unaffected (or very slightly—<5% faster) by the change, CPU-intense operations often get a 20-30% speedup. So at least for some applications, getting to 64-bit would be great.

Obviously, there would either be a bit of work getting parallel 32/64-bit builds going, or you'd have to cut off all the 32-bit-only users (anyone not on a Pi 2 model B or newer, Pi Zero excluded) :(