hypriot/image-builder-rpi #366: Issue with setting up eth0 with static ip address

dennym, 2020-06-12:
I gave Hypriot-1.12.2 and run today and configured cloud-init config and everything got flashed and initialized correctly with your provided flash tool.

Unfortunately the static ip address for eth0 does and won't work. I tried the example from  https://github.com/hypriot/flash/tree/master/sample/static.yml .

```
write_files:
  - content: |
      persistent
      slaac private
      interface eth0
      static ip_address=192.168.0.31/24
      static routers=192.168.0.1
      static domain_name_servers=192.168.0.1 8.8.8.8
    path: /etc/dhcpcd.conf
```

After the PIs are up and running I checked via ssh if the file exists and `/etc/dhcpcd.conf` exists with the mentioned content. Since it wasn't applied to the network I though about restarting the service with `sudo service dhcpcd restart` but this service is masked and can't be actually used. As far I am aware if something is masked its unused and dead. Since the example are quite old I guess a lot of stuff changed since then. Is there another approach for this?

best

rbartoli, 2020-06-16:
I reported same issue in #359 and I've still not found the solution.

p-hagen-Signalfx, 2020-07-20:
In the root of your SD card, there is a file called network-config
this is where you set the settings that get used during cloud-config
This is what is in mine, and this works:

```
version: 1
config:
  - type: physical
    name: eth0
    subnets:
      - type: static
        address: 192.168.1.10/24
        gateway: 192.168.1.1
  - type: nameserver
    address:
      - 192.168.1.1
```

Bazmundi, 2020-08-20:
> In the root of your SD card, there is a file called network-config
> this is where you set the settings that get used during cloud-config
> This is what is in mine, and this works:
>
> ```
> version: 1 [...]

> In the root of your SD card, there is a file called network-config
> this is where you set the settings that get used during cloud-config
> This is what is in mine, and this works:
>
> ```
> version: 1 [...]

Not a lot of help, people are using the flash installer and following the instructions to edit the yml file from a supplied example.   So there is something broken with the config setup.

Bazmundi, 2020-08-20:
> Hypriot-1.12.2

Same for Hypriot-1.12.3

Bazmundi, 2020-08-20:
So @dennym and @rbartoli I have the same problem against Hypriot-1.12.3.  I used an older 2.4.0 of Flash, not the latest 2.7.0.  I am not sure if that will help but I will try to reflash after updating Flash.

Bazmundi, 2020-08-23:
> As far as I can tell the lates version of Hyprio introduced a change on how network is set following the cloud-init best practice. in short at boot it reads the _network-config_ that you can find in the root folder and applies it content to your system. You need to edit this before you boot the Pi for the firs time.
> 
> Now, if you change the content with the above example he system will set a static ip address with the info for **MY** network, applied for eth0, and it allows me to use docker-compose up without any problems.
> 
> ```
> $ docker-compose up
> Creating network "composetest_default" with the default driver
> Building web
> Step 1/10 : FROM python:3.7-alpine
> 3.7-alpine: Pulling from library/python
> 52278dd8e579: Pull complete
> 9e521c9bf15d: Pull complete
> 95f2a107cc48: Pull complete
> 39a5fb8cc132: Pull complete
> b70cfcc5b706: Pull complete
> Digest: sha256:d1375bf0b889822c603622dc137b24fb7064e6c1863de8cc4262b61901ce4390
> Status: Downloaded newer image for python:3.7-alpine
>  ---> 84eaaf2db0c5
> Step 2/10 : WORKDIR /code.........
> ```
> 
> You must change it to match **_your_** own network configuration for it to work, but using network-config to set your systems network before boot seems the way forward
> 
> you can find info on cloud-init and networking here https://lxd.readthedocs.io/en/stable-2.0/cloud-init/

Affirmative. You are correct. The setup examples are out of date.

highvight, 2020-08-31:
@Bazmundi Perhaps this solution with flash helps you on the [related issue](https://github.com/hypriot/image-builder-rpi/issues/359#issuecomment-683513795)