hypriot/image-builder-rpi #368: Windows Support

zwhitchcox, 2020-07-11:
Is this project supposed to support windows? there is no output when I run `make`

zwhitchcox, 2020-07-11:
I mean, no file output

zwhitchcox, 2020-07-11:
Ok, now on `make sd-image` I'm getting this error:

```
docker run --rm --privileged -v "/c/Users/zwhit/dev/cluster-pi/image-builder-rpi:/workspace" -v "/boot:/boot" -v "/lib/modules:/lib/modules" -e CIRCLE_TAG -e VERSION image-builder-rpi
C:\Program Files\Docker Toolbox\docker.exe: Error response from daemon: invalid mode: \Program Files\Git\workspace.
```

zwhitchcox, 2020-07-12:
Ok, now even on ubuntu 18.04, I'm getting this error:

```
Get:4 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-sys-mods armhf 20200514 [10.6 kB]
Get:5 http://archive.raspberrypi.org/debian buster/main armhf pi-bluetooth all 0.1.13 [4908 B]
Get:6 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf libdw1 armhf 0.176-1.1 [218 kB]
Get:7 http://mirror.pit.teraswitch.com/raspbian/raspbian buster/main armhf libglib2.0-0 armhf 2.58.3-2+deb10u2 [1078 kB]
debconf: unable to initialize frontend: Dialog
debconf: (TERM is not set, so the dialog frontend is not usable.)
debconf: falling back to frontend: Readline
debconf: unable to initialize frontend: Readline
debconf: (This frontend requires a controlling tty.)
debconf: falling back to frontend: Teletype
dpkg-preconfigure: unable to re-open stdin: 
Fetched 2314 kB in 2s (994 kB/s)
Selecting previously unselected package gettext-base.
(Reading database ... 25647 files and directories currently installed.)
Preparing to unpack .../0-gettext-base_0.19.8.1-9_armhf.deb ...
Unpacking gettext-base (0.19.8.1-9) ...
Selecting previously unselected package libdw1:armhf.
Preparing to unpack .../1-libdw1_0.176-1.1_armhf.deb ...
Unpacking libdw1:armhf (0.176-1.1) ...
Selecting previously unselected package libglib2.0-0:armhf.
Preparing to unpack .../2-libglib2.0-0_2.58.3-2+deb10u2_armhf.deb ...
Unpacking libglib2.0-0:armhf (2.58.3-2+deb10u2) ...
Selecting previously unselected package bluez.
Preparing to unpack .../3-bluez_5.50-1.2~deb10u1+rpt1_armhf.deb ...
Unpacking bluez (5.50-1.2~deb10u1+rpt1) ...
Selecting previously unselected package bluez-firmware.
Preparing to unpack .../4-bluez-firmware_1.2-4+rpt3_all.deb ...
Unpacking bluez-firmware (1.2-4+rpt3) ...
Selecting previously unselected package raspberrypi-sys-mods.
Preparing to unpack .../5-raspberrypi-sys-mods_20200514_armhf.deb ...
Unpacking raspberrypi-sys-mods (20200514) ...
Selecting previously unselected package pi-bluetooth.
Preparing to unpack .../6-pi-bluetooth_0.1.13_all.deb ...
Unpacking pi-bluetooth (0.1.13) ...
Setting up bluez-firmware (1.2-4+rpt3) ...
Setting up libdw1:armhf (0.176-1.1) ...
Setting up libglib2.0-0:armhf (2.58.3-2+deb10u2) ...
No schema files found: doing nothing.
Setting up gettext-base (0.19.8.1-9) ...
Setting up raspberrypi-sys-mods (20200514) ...
Enabling audio in config.txt and removing from /etc/modules...
qemu: Unsupported syscall: 382
Fixing file capabilities...
Failed to set capabilities on file `/bin/ping6' (Invalid argument)
The value of the capability argument is not permitted for a file. Or the file is not a regular (non-symlink) file
Adjusting cmdline.txt...
Removing old sysctl tweaks, if present...
Removing old NOPASSWD line, if present...
Updating documentation URL in /boot/config.txt...
Updating apt sources from {mirrordirector,archive}.raspbian.org to raspbian.raspberrypi.org...
Created symlink /etc/systemd/system/reboot.target.wants/rpi-display-backlight.service → /lib/systemd/system/rpi-display-backlight.service.
Created symlink /etc/systemd/system/halt.target.wants/rpi-display-backlight.service → /lib/systemd/system/rpi-display-backlight.service.
Created symlink /etc/systemd/system/poweroff.target.wants/rpi-display-backlight.service → /lib/systemd/system/rpi-display-backlight.service.
Created symlink /etc/systemd/system/multi-user.target.wants/sshswitch.service → /lib/systemd/system/sshswitch.service.
Setting up bluez (5.50-1.2~deb10u1+rpt1) ...
Running in chroot, ignoring request.
Reloading system message bus config...Failed to open connection to "system" message bus: Failed to connect to socket /var/run/dbus/system_bus_socket: No such file or directoryinvoke-rc.d: initscript dbus, action "force-reload" failed.
Created symlink /etc/systemd/system/dbus-org.bluez.service → /lib/systemd/system/bluetooth.service.
Created symlink /etc/systemd/system/bluetooth.target.wants/bluetooth.service → /lib/systemd/system/bluetooth.service.
Running in chroot, ignoring request.
Setting up pi-bluetooth (0.1.13) ...
Created symlink /etc/systemd/system/multi-user.target.wants/hciuart.service → /etc/systemd/system/hciuart.service.
Processing triggers for systemd (241-7~deb10u4+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
qemu: Unsupported syscall: 383
qemu: Unsupported syscall: 383
qemu: Unsupported syscall: 383
[... 889 lines cut ...]
        cmd_obj.run()
      File "/tmp/pip-build-env-j6xtd9kn/lib/python3.7/site-packages/setuptools/command/install.py", line 61, in run
        return orig.install.run(self)
      File "/usr/lib/python3.7/distutils/command/install.py", line 589, in run
        self.run_command('build')
      File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
        self.distribution.run_command(command)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/usr/lib/python3.7/distutils/command/build.py", line 135, in run
        self.run_command(cmd_name)
      File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
        self.distribution.run_command(command)
      File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
        cmd_obj.run()
      File "/tmp/pip-install-kjtdq478/pynacl/setup.py", line 161, in run
        raise Exception("ERROR: The 'make' utility is missing from PATH")
    Exception: ERROR: The 'make' utility is missing from PATH

[... cut ...] ------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-kjtdq478/pynacl/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-o0hf2moz/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-kjtdq478/pynacl/
Makefile:7: recipe for target 'sd-image' failed
make: *** [sd-image] Error 1
```

schnAlex, 2020-07-22:
same error on macos. docker, vagrant and virtualbox are up-to-date.

QuantumLove, 2020-08-05:
Same with Ubuntu 20.04