hypriot/image-builder-rpi #381: current time is set too late during cloud-init

jhofmueller, 2021-01-15:
I am aware that this issue has been raised several times in different contexts here.  Nevertheless I have to open a new issue since the problem either still exists or has come back.

My recent attempts to setup an rpi4 and install packages using the cloud-init way failed for well known reasons.  When cloud-init starts it prints this line:

`from /var/log/cloud-init-output.log:
Cloud-init v. 18.3 running 'init-local' at Thu, 14 Feb 2019 10:12:03 +0000. Up 6.96 seconds.`

Which results in errors like this:

`from /var/log/cloud-init-output.log:
Certificate verification failed: The certificate is NOT trusted. The certificate chain uses not yet valid certificate.  Could not handshake: Error in the certificate verification. [IP: 13.32.2.103 443]`

and finally in the failure of apt with a most curious log line:

`from /var/log/cloud-init.log:
2021-01-15 10:54:06,802 - util.py[DEBUG]: apt-update [apt-get --option=Dpkg::Options::=--force-confold --option=Dpkg::options::=--force-unsafe-io --assume-yes --quiet update] took 60568897.719 seconds`

Sometime while apt was running the time was correctly set, but that happened too late.

I checked the `/etc/fake-hwclock.data` which links to `/boot/fake-hwclock.data`.  The timestamp is current.

Regards,
j.

psdu72, 2021-01-23:
Seems that the project is not being updated since almost 7 months now, I am asking myself if it makes sense to give it a try or not...

jhofmueller, 2021-01-25:
Opening this issue is also kind of a ping to see if the project will be maintained in the future or not :smirk: