hypriot/image-builder-rpi #385: Has this project been abandoned?

leonheess, 2022-02-05:
I want to restructure my home network and Hypriot sounds like the perfect thing but it seems unmaintained - is it?

letsjustfixit, 2022-04-11:
=(

illegitimate-egg, 2022-07-27:
I think so, they removed the donation link from the readme

greenonline, 2023-04-12:
It is such a shame that it has been abandoned, especially after Jeff Geerling has hyped it up in his CM cluster videos. He mentions [here](https://www.reddit.com/r/homelab/comments/s0b7gk/comment/hvrya3i/?utm_source=share&utm_medium=web3x) that he might do a new set of videos not using HyperiotOS:

> I should probably do another cluster series soon; the Hyperiot OS choice wasn't that amazing since they kind of slowed/stopped developing it. I use Pi OS now for my Pi work, as it is 64 bit and almost identical to Debian.

gounthar, 2023-04-15:
Armbian works well with Pi3B+ and 4 now.